nauful/LibUA #4: .net core support

alenloncaric, 2019-10-22:
Is there a possiblity to support .net core?

nauful, 2019-10-27:
Hello,

I last looked at this possibility ~3 years ago and found that .NET Core, at that time, was missing many required security libraries from normal .NET. While I would guess that support is now available, or equivalents can be found, I don't have time to develop anything further in OPC UA.

If anyone else would like to attempt this, check that symmetric and asymmetric crypto methods for signing and encryption work.

Regards,
Nauful

nauful, 2020-03-07:
Closing for now, no immediate plans to support .NET Core. This may happen in the future if I return to the world of OPC UA and Scada systems.

closed 2020-03-07 by nauful

gogolevsergey, 2020-03-17:
#7 resolve this

nauful, 2020-03-18:
Merged: https://github.com/nauful/LibUA/tree/336611f534365c4c52e4ba966f7f62bdf22e45e2