nauful/LibUA #6: UaExpert client not connect

Polyteres, 2020-03-06:
Hi, I have a problem using TestServer and trying to connect the UaExpert client. I use the TestServer example provided in the repository as it is and I can't connect to that client. I would appreciate if you could show me how to configure the connection for that client or if you could explain to me the new ApplicationDescription () function.

Thank you.

nauful, 2020-03-07:
Hello,

Quick sanity checklist: 
Are you running the static method named TestServer()
If so, try to add the server 127.0.0.1:7718 in UaExpert. Does UaExpert discover endpoints with varying security options?
Either way, set a breakpoint on GetEndpointDescriptions and check that UaExpert is connecting to enumerate the endpoint options (breakpoint will be hit).
If so, try connecting to the one that says None in UaExpert (security policy None).  Set a breakpoint on sessionValidateClientApplication in the sample application, does it get there?
If so, you can additionally try the other security modes (their connection will always be accepted because of return true in sessionValidateClientApplication).

The ApplicationDescription structure is just some info to show UA clients what your UA server's purpose is. The GetApplicationDescription method can return a different ApplicationDescription depending on the URL queried by a client.

Polyteres, 2020-03-10:
Hello,

> Are you running the static method named TestServer()

Yes.

> If so, try to add the server 127.0.0.1:7718 in UaExpert. Does UaExpert discover endpoints with varying security options?

No.

I set a breakpoint on GetEndpointDescription but it never comes. I attach a screenshot:

https://www.dropbox.com/s/i2e92uk1m229y3s/OPCUA.jpg?dl=0

Thank you.

nauful, 2020-03-10:
It seems that the request for endpoints is failing, which works for me locally.

You should try setting a breakpoint on these two methods and stepping through:
https://github.com/nauful/LibUA/blob/019448e5f34f9f41d62196ff891ef1bae63fa284/LibUA/NetDispatcher.cs#L1097
https://github.com/nauful/LibUA/blob/019448e5f34f9f41d62196ff891ef1bae63fa284/LibUA/NetDispatcher.cs#L1058

These two message types can be handled without an established session and are used for UA discovery.

Polyteres, 2020-03-11:
Hello, I have set a breakpoint in those functions but nothing happened. Please, Could you try UAExpert client with your server and make sure it works, and if so, give me all the steps to follow?. I'm desperate because nothing works for me.

Thank you very much.

nauful, 2020-03-11:
This is very strange because when I run TestServer, I am able to connect fine with UaExpert. I am running uaexpert-bin-win32-x86-vs2008sp1-v1.3.0-201 locally.

I just add opc.tcp://127.0.0.1:7718 under Custom Discovery and the endpoints are listed properly.

Could you run UaExpert on one computer and your server on another, and use Wireshark to capture traffic on either end? Zip and attach the captured traffic here. This will show me the messages sent by UaExpert and the library's responses so I can look for any issues. NetDispatcher handles incoming messages from connected clients, so if those are not being triggered then something is definitely wrong.

loongtail, 2020-03-12:
Hello, I came across the same problem, and I found my SecurityPolicy is "Invalid" when I set a breakpoint on 
`protected int DispatchOpen(SLChannel config, MemoryBuffer recvBuf)`

I'm using UaExpert 1.5.1 331, but I'm not sure it is the reason.

Polyteres, 2020-03-12:
Hello, I just checked it and I have the same problem as @loongtail.

nauful, 2020-03-12:
The default security policy should be http://opcfoundation.org/UA/SecurityPolicy#None
This remains until the session is created and established, when a potentially different security policy can be chosen.

If UaExpert expects something different now, I'd like to find out what it's doing. Could you please capture the Wireshark traffic?

Polyteres, 2020-03-12:
Hello, I have tried version 1.2 of UAExpert and I did not find any problem either. I have seen that you have used version 1.3 of UAExpert. I have had problems with later versions, 1.5 (@loongtail) and 1.4.4 (me) which is the one I am currently using. I think if you directly download the latest version of UAExpert you will find the same problems that we have had. This problem that I have mentioned also happens to us with another SCADA of a client so it is not an isolated case or referred only to UAExpert.

Best regards.

loongtail, 2020-03-12:
@nauful  I'm not sure whether it's the data you want, I just filtered the data by IPs of two ends.
[UA_WS.zip](https://github.com/nauful/LibUA/files/4324579/UA_WS.zip)

nauful, 2020-03-12:
I only see the connection request in this Wireshark capture, not the response.

I don't have any insights into what's changed in UaExpert as a client, it's sending the same OpenSecureChannel message as before. Perhaps it now expects something differently, which breaks compatibility with most existing UA servers.

loongtail, 2020-03-12:
I don't know how to use the wireshark correctly, maybe I filtered by wrong key, whatever, thank you for your support :)

nauful, 2020-03-12:
Happy to help, I just don't see anything that would cause an issue being returned. I don't have an account to download new OPC Foundation tools, will investigate further when I get the chance to.

loongtail, 2020-03-13:
[WS&UAExpert1.51.zip](https://github.com/nauful/LibUA/files/4327308/WS.UAExpert1.51.zip)
I tried filtering again the traffic, and packed it with the UAExpert1.51, hope to help your investigation.

nauful, 2020-03-13:
[image]

Very helpful, thanks. In the Wireshark capture I can see that a valid response was sent for GetEndpointsResponse, so the issue seems localized to the UaExpert client library (OPC Foundation's client library). I'll install UaExpert 1.5.1 and take a closer look.

nauful, 2020-03-13:
I have found one bug in the OPC Foundation's current client library: maximum chunk size of 64k causes an exception (their client tries to read 65536 from a buffer of 65535). As a workaround (for compatibility... a never-ending game of whack-a-mole), I am updating DispatchHello with this local config for transmission limits:

```
const uint chunkSize = (1 << 16) - 1; // Was previously 1 << 16 (65536, now 65535)
config.TL.LocalConfig = new TLConfiguration()
{
	ProtocolVersion = 0,
	SendBufferSize = chunkSize,
	RecvBufferSize = chunkSize,
	MaxMessageSize = (uint)maximumMessageSize,
	MaxChunkCount = (uint)(maximumMessageSize + (chunkSize - 1)) / chunkSize,
};
```

UaExpert 1.5.1 still has an issue, continuing to investigate.

Polyteres, 2020-03-13:
Thank you very much for your support. We await more news from you. Thanks again.

nauful, 2020-03-13:
UaExpert 1.5.1 seems to have raised the minimum acceptable token lifetime, so changing the default from 1 minute to 10 minutes has fixed the problem. Updated source with two commits, one for NetDispatcher with the change above and one in Server for this:

```
public const int MaxTokenLifetime = 600 * 1000;
```

[image]
[image]

Please take a look and let me know if it works, then you can close the issue.

Larry57, 2020-03-13:
Hi, I was just trying UAExpert 1.5.1.311 too at the same time you commited changes : now it is ok on my side despite of the messages that pops up in the log window.
Thanks a lot for this code, I need to study the server part now and give it a try.

loongtail, 2020-03-13:
Thanks a lot, it works now.

Polyteres, 2020-03-13:
Hello, thanks a lot!!!. It works now, yes. Thank you very much for your support. I close the issue, thanks again.

closed 2020-03-13 by Polyteres

nauful, 2020-03-13:
Welcome, feel free to ask if you have any questions or other problems.