nauful/LibUA #8: I do not understand these errors

Polyteres, 2020-03-31:
Hi again. Using the server example I think everything updates fine but I get these errors in the UAExpert console.  What do they mean ?. Is everything working correctly ?.

https://www.dropbox.com/s/xfkvd85kxi1m5hc/error.jpg?dl=0

Thank you.

nauful, 2020-04-01:
It seems that there is an issue with publishing, however I was able to subscribe to tags and see them in the middle window without issue.

What steps did you take after connecting to reach those errors?

Polyteres, 2020-04-02:
Hi, I just conneted and all those errors started. I didn't do anything else. Here are all the steps I have done:
- Create a project in Visual Studio 2019 with .NET 4.7.2.
- Add the folder structure to the project (Add Folder).
- Copy the files into the created folder structure.
- Add the files with Add existing item.
- Fix the references (only one).
- Add the -unsafe flag to the project to be able to compile.
- In Security.cs I avoid the problem with the names using LibUA.Security.Cryptography.RSACng () in the two errors that I get.
- Copy the contents of Test.erver Program.cs to my Program.cs of my project.
- Fix the naming problem using LibUA.Security.Cryptography.RSACng ()
- Execute.
- Allow access to the windows firewall.
- Open UAExpert 1.5.1
- Looking for the server in opc.tcp: //127.0.0.1: 7718
- Select the server using None
- Connect to the server
- Errors appear continuously.

Thank you.

nauful, 2020-04-02:
Was it working fine earlier? Do you see errors if connecting with signed and encrypted security?

I'll try to reproduce those steps but for now here are my two project directories from VS2015 along with binaries in the Debug directory to test with. TestServer.zip contains what you're using.

[TestClient.zip](https://github.com/nauful/LibUA/files/4421137/TestClient.zip)
[TestServer.zip](https://github.com/nauful/LibUA/files/4421138/TestServer.zip)

Polyteres, 2020-04-03:
Hi, mistakes have always been present. With signed an encrypted security everything remains the same.

Polyteres, 2020-04-03:
Hi, again. Like the query I made a few weeks ago, when it won't connect I don't know if you remember, it was suspicious of the UAExpert version. I have downloaded version 1.2 of UAExpert and tested TestServer. Everything works correctly without any type of error, so I suspect that something changed in later versions of UAExpert that causes those errors to appear just as it happened with the connection error, already solved.

Thanks you for you support.

nauful, 2020-04-03:
Could you please share the installer for the client that you're using (upload .zip or link .7z)?

I can install it locally and test.

Polyteres, 2020-04-03:
Hi, of course:

- TestServer VS 2019 Project: https://www.dropbox.com/s/6y9wgixz2c99qkj/LibUAClear.rar?dl=0

- UAExpert 1.5.1: https://www.dropbox.com/s/65t6qqp534dfqg7/uaexpert-bin-win32-x86-vs2008sp1-v1.5.1-331.zip?dl=0

Thank you.

nauful, 2020-04-04:
Using the normal .NET version, not .NET Core, I don't see any of the errors. If you're using the new .NET Core version, maybe expected behaviour in .NET has changed and I can investigate that.

For now, can you test with the TestServer.zip that I've uploaded? It's the same source as the repo, working fine right now.

Ernestopv, 2020-04-07:
Hi does this stack supports reverse hello ?

nauful, 2020-04-07:
No, it doesn't natively support OPC UA reverse hello and gateways beyond the normal single-server profile. If you want to try adding support, look at Server.cs and NetDispatcher.cs in Server.

nauful, 2020-04-13:
@Polyteres Do you see the issue with the zip I uploaded, or the current version of the source, or neither?

I tested locally but found that it worked fine.

nauful, 2020-04-16:
Could not reproduce, please re-open if the issue persists.

closed 2020-04-16 by nauful