nauful/LibUA #11: LibUA only supports obsolete security policies

Ttberg, 2020-04-08:
According to the OPC foundation, the security policies Basic256 and Basic128Rsa15 are obsolete. Is there any plan to implement e.g. Basic256Sha256?

nauful, 2020-04-08:
Support will likely be added for this in the future but I don't have time at the moment to resume development beyond bug fixes and support. I'd invite anyone with the time to investigate, implement and test to create a pull request that implements new security policies. Right now the only implemented policies are None, Basic256 and Basic128Rsa15.

Security.cs contains implementations for the signing and encryption schemes if you'd like to try adding support.

Closing the issue for now, will re-open and resolve when support is added, either via contributor or when I get time to investigate.

closed 2020-04-08 by nauful

nauful, 2020-04-20:
Started adding support for Basic256Sha256 as a security policy:
https://github.com/nauful/LibUA/tree/prototype_security_basic256sha256