nauful/LibUA #12: How to read a single node id?

hulinning2, 2020-04-14:
Hi, 
Sorry, I am so new to OPCUA and I only need to read a couple's data points using Node ID.
I am using a demo version of Kepware OPC UA Client and all I have to do is specified the server IP address and port number. After that, I can read any node id.
I am connecting to the server without any security at this point.

Could you point out which functions that allow me to read a single node id in your LibUA?

Thanks a lot

nauful, 2020-04-14:
Are you connecting to LibUA server via an OPC UA client?

On the server side, to handle a node read request, look at
https://github.com/nauful/LibUA/blob/4c9e1e7a79856d436523227ee206fdf1394d5f61/NET/TestServer/Program.cs#L185

I minimally defined some required OPC UA nodes in ns = 0, but you can return anything you want in a user-definable namespace (ns >= 2). This is a very light weight model, so you don't have to create or store anything anywhere, just return values according to requests. The same method handles single and multiple read requests, so you don't have to implement a different method for each of the UA read options.

hulinning2, 2020-04-14:
Hi Nauful, 
I just need to use your LibUA client to read the value of a node id on different OPC UA server.
I try to figure out how to connect to the OPC UA server without security and read a value of node id at any point in time. 
Does OPC UA have any time out on the server or client?
Thanks for your help

nauful, 2020-04-14:
OPC UA client has a timeout specified here:
https://github.com/nauful/LibUA/blob/4c9e1e7a79856d436523227ee206fdf1394d5f61/NET/TestClient/Program.cs#L113
The operational timeout is min(client timeout, server timeout). If the read call, or any other call fails with timeout, the return status code is BadRequestTimeout.

Here you can see an example of reading a node on the server. I read three node ids with one call (batched):
https://github.com/nauful/LibUA/blob/4c9e1e7a79856d436523227ee206fdf1394d5f61/NET/TestClient/Program.cs#L142

hulinning2, 2020-04-14:
Thanks a lot

closed 2020-04-14 by hulinning2