nauful/LibUA #16: Monitored item delivers errors in clients log.

antroa, 2020-05-06:
Monitoring "Item\Trend 000001" with UAExpert delivers every second this log-messages in UAExpert's log output:

```
Error: Number of acknowledgement results (0) does not match number of acknowledgements (2)
Error: UaSessionPrivate::publishCallback - List of available sequence numbers is empty, at least current sequence number has to be available
```

nauful, 2020-05-07:
Fixed: https://github.com/nauful/LibUA/commit/4a2ee9a48438d8aceeca55840320a05c2cb440a6

nauful, 2020-05-07:
How do you reproduce this? Acknowledgements are being sent back, perhaps a configuration issue that I can investigate.

[image]

nauful, 2020-05-14:
Closing, could not reproduce.

closed 2020-05-14 by nauful