nauful/LibUA #20: Missing type information for method calls.

antroa, 2020-05-06:
The uaexpert client can show input- and output value fields build from type information supported by the method node. Did libua demo server support these informations?

nauful, 2020-05-07:
You could add this by returning appropriate type information in the read node method, but this isn't included by default due to increased memory requirements to store that information. The spec is lax in terms of definitions.

closed 2020-05-07 by nauful

antroa, 2020-05-08:
Thank you for your answer (also regarding my other questions). Yes, the spec. is lax. Therefore, the CTT exists to clarify the question "what is conform", more than the spec., but of course not 100%.
Have you plans to test libua with CTT?

nauful, 2020-05-08:
No, most vendor implementations don't fit the CTT, and changes in the OPC Foundation stack often break the CTT in contradicting ways.

I usually test against several major vendors (OPC Foundation, Yokogawa, Prosys, some smaller ones that our company's clients use). I aim for general wide-spread compatibility rather than rare features which aren't otherwise necessary, works very well for most use cases.