nauful/LibUA #22: Matrikon OPC UA Explorer client not connect

FieryPhoenix100, 2020-05-18:
Hi, I try to connect to Matrikon OPC UA Explorer but it not find endpoints. I get following error in  Matrikon OPC UA Explorer console: No endpoints available for URL.
I use Net Core\TestServer.

nauful, 2020-05-20:
Could you please provide a link or attach the client executable or installer?

FieryPhoenix100, 2020-05-20:
You can download the installation file here: https://www.matrikonopc.com/downloads/1426/downloads/index.aspx

nauful, 2020-05-20:
It requires registration, could you attach the exact version that you're using?

FieryPhoenix100, 2020-05-20:
Can you download the installer from my dropbox: https://www.dropbox.com/s/7ukctnopnck2e11/Matrikon_OPC_UA_Explorer.msi?dl=0?
Unfortunately I can't attach the installer to the comment due to the github limit of 10 mb.

nauful, 2020-05-20:
Looks good, I'll investigate this shortly and get back to you. Sorry, due to the current volume of work it will take a bit of time.

nauful, 2020-05-23:
[image]

I haven't implemented the FindServer call in the demo server because this is a gateway method, not a normal OPC UA server method. While I might implement this in the future, for now you'd be better off with a gateway server to discover OPC UA endpoints on a system.

GetEndpoints is implemented for the current server, but FindServer is for gateway servers to direct an incoming client to another server. Closing for now.

closed 2020-05-23 by nauful