nauful/LibUA #25: Problem with item monitor with Prosys OPC UA Client

Polyteres, 2020-06-17:
Hi, I have problems with Prosys OPC UA Client. Using the provided server example, when I make a subscription (monitor) to a variable, it doesn't show any data (value).

[Capture with the problem.](https://www.dropbox.com/s/4pr8rroe9kd68wv/problem.jpg?dl=0)

Download Prosys OPC UA Client from my Dropbox [here](https://www.dropbox.com/s/6jzmacuibr0c3mk/prosys-opc-ua-client-3.0.0-256.exe?dl=0).

Thanks for you support.

nauful, 2020-06-18:
How do you connect with this client? I have clicked on the right arrow icon (tooltip: connect to server) but nothing happens in the client.

[image]

Server prints the following:
```[Info] Accepted connection from 127.0.0.1:49462
[Info] 127.0.0.1:49462: SL security token 1000 issued for channel 1 with security policy None
[Info] 127.0.0.1:49462: Message type ns=0;i=422 with SL state Opening
[Info] 127.0.0.1:49462: Message type ns=0;i=452 with SL state Opening
[Info] 127.0.0.1:49462: Client sent CloseSecureChannelRequest
[Info] Ended connection from 127.0.0.1:49462
[Info] Accepted connection from 127.0.0.1:49463
[Info] 127.0.0.1:49463: SL security token 1000 issued for channel 1 with security policy None
[Info] 127.0.0.1:49463: Message type ns=0;i=422 with SL state Opening
[Info] 127.0.0.1:49463: Message type ns=0;i=452 with SL state Opening
[Info] 127.0.0.1:49463: Client sent CloseSecureChannelRequest
[Info] Ended connection from 127.0.0.1:49463
[Info] Accepted connection from 127.0.0.1:49464
[Info] 127.0.0.1:49464: SL security token 1000 issued for channel 1 with security policy None
[Info] 127.0.0.1:49464: Message type ns=0;i=428 with SL state Opening
[Info] 127.0.0.1:49464: Message type ns=0;i=452 with SL state Opening
[Info] 127.0.0.1:49464: Client sent CloseSecureChannelRequest
[Info] Ended connection from 127.0.0.1:49464
[Info] Accepted connection from 127.0.0.1:49465
[Info] 127.0.0.1:49465: SL security token 1000 issued for channel 1 with security policy None
[Info] 127.0.0.1:49465: Message type ns=0;i=428 with SL state Opening
[Info] 127.0.0.1:49465: Message type ns=0;i=452 with SL state Opening
[Info] 127.0.0.1:49465: Client sent CloseSecureChannelRequest
[Info] Ended connection from 127.0.0.1:49465
```

nauful, 2020-06-18:
The client never seems to send any publish requests, and data change notifications are only set once per publish requests to throttle traffic.

Could you also check this against a different UA server? I would like to check what triggers this client into sending a publish request, because playing around with the subscription settings hasn't gotten any requests sent.

Polyteres, 2020-06-18:
Hello, I have tried it with opcuaserver.com:48010 with security mode and security policy selected to None, and after waiting a long time for it to connect, I have been able to subscribe to an item and it has worked. 

[Image with subscription working](https://www.dropbox.com/s/mndv91tt2kh78wr/prosys.jpg?dl=0)

Thank you for you support.

nauful, 2020-06-19:
I seem to be returning the same responses, verified with Wireshark... wonder if it's something to do with floats that the client does not send publish requests. I suspect a bug or misconfiguration in the client, will continue to investigate why it sends a publish request to opcuaserver.com and not this server.

nauful, 2020-06-22:
Wireshark capture:
[captured.zip](https://github.com/nauful/LibUA/files/4814141/captured.zip)

The client never sends a PublishRequest message, so no data is published. Do you have source access to this client, or any client that I can step through to investigate why? There isn't a problem on the server side. I'm looking for a workaround to satisfy this client.

Polyteres, 2020-06-23:
Hi, unfortunately I don't have access to the sources, but I'm going to look for some solution that can make it easier for you to find the error.

Polyteres, 2020-07-06:
Hi nauful!. I don't want to be intense but have you found any solution?. 

Thanks.

nauful, 2020-07-06:
No, I haven't been able to find a workaround. I don't see any issue with the server side because there are no errors. I'll investigate the client once more when I have time.

If you have source for a client that has the same issue, I can look for what the client misses.

nauful, 2020-07-08:
Modified log4j.properties in prosys-opc-ua-client-app-3.0.0-256.jar to enable trace-level logging.
[logger_ua.zip](https://github.com/nauful/LibUA/files/4893406/logger_ua.zip)

I do not see any errors nor exceptions in the trace-level log, not sure why it will not send publish requests. Closing this issue for now until I can find more information for a client-specific workaround.

I suspect mimicking more nodes within the default UA address space might make this work over time at some point. If you would like to try, the next step would probably be an ExtensionObject of ServerStatusDataType for ns=0;i=2256.

closed 2020-07-08 by nauful