nauful/LibUA #27: UaExpert Error: Number of acknowledgement results (0) does not match number of acknowledgements(1)

davidguidali, 2020-07-10:
Hi 
Thank you very much for your client / server implementations, I like them a lot!
I am actually learning OPC-UA, so this may be a basic question. I tried your server and connected with UA Expert. Everything seems to work fine, however I get this error a lot in the logs:

Error: Number of acknowledgement results (0) does not match number of acknowledgements(1)

[image]

Do you know whats the issue there?

mweissi, 2020-07-10:
I also observe this messages with UaExpert

nauful, 2020-07-10:
Could you please upload an installer for your version of UaExpert?

I suspect behaviour is different from the previous version(s) now:
[image]

davidguidali, 2020-07-10:
Thats the client I'm using, downloaded from unified automation page https://www.unified-automation.com/de/downloads/opc-ua-clients/file/download/details/uaexpert-v144.html

[uaexpert-bin-win32-x86-vs2008sp1-v1.5.1-331.zip](https://github.com/nauful/LibUA/files/4904342/uaexpert-bin-win32-x86-vs2008sp1-v1.5.1-331.zip)

nauful, 2020-07-10:
Done, added support for subscription acknowledgements and publishing available sequence numbers.

Note that subscription acknowledgements are asynchronous, so you might still see a harmless warning (e.g. sent 0 or 2 when 1 is consistently returned from the server). Available sequence numbers only cover the current published data change or event notification.

closed 2020-07-10 by nauful

davidguidali, 2020-07-11:
Cool, thanks very much! :)