nauful/LibUA #32: Connect to Server problem

NikolayStek, 2020-08-18:
Hello nauful,

Thank you for your work!

I have some problem with OpenAsync with UA Automation SDK Client Example(https://www.unified-automation.com/news/news-details/article/new-net-based-opc-ua-clientserver-sdk-v305-service-release.html) and Convertersystems\opc-ua-client (https://github.com/convertersystems/opc-ua-client)

Endpoint Discover is Ok, but then I chose Endpoint and try to connect:
Then I try **_None_** secure connection, I have 1-3 messages like: _[Info] 127.0.0.1:13813: Message type ns=0;i=428 with SL state Opening_ and then _[Info] Ended connection from 127.0.0.1:14290_.

Then I try _**Basic256Sha256**_ secure connection, I have 1-3 messages like: _[Info] 127.0.0.1:13813: Message type ns=0;i=428 with SL state Opening_ and then _[Error] Sent TL error 0x80070000_, _[Info] Ended connection from 127.0.0.1:14290_.

If you have some time can you check? For save time you can use opc-ua-client, they have Unit Tests -> IntegrationTests.cs -> ConnnectToAllEndpoints().

nauful, 2020-08-18:
I'm not a member on that site and they haven't approved my registered account. Could you upload the installers and link here?

Alternatively, a WireShark capture of network traffic would help. One capture with None and another with Basic256Sha256.

NikolayStek, 2020-08-19:
Thank's for fast response.

Here a SDK and WireShark capture of network traffic with None and another with Basic256Sha256 for UA SDK and Convertersystems\opc-ua-client. https://dropmefiles.com/KnqmX

During the trafic capture I found that 
1. Convertersystems\opc-ua-client - get server endpoints correct but can't connet to any of them
2. UA SDK Client - get server endpoints but during that procese server write _[Error] Sent TL error 0x80070000_, 
client can't connect with None (I think that the problem is that the your server not send his certificate in None secure connection),
client can connect with any secure, but has a problem with Disconnection

Also I found meybe other issue: Your client can't connet to UA SDK Server - If it's interesting i can make WireShark capture

nauful, 2020-08-19:
Looks like an issue with certificate trusting. 

Client to UA SDK server (applies to with and without security):
Either trust the client certificate or enable this line in the .xml config: <AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>
Otherwise the client will be rejected with BadSecurityChecksFailed when opening a session.
[image]
After trusting (copied sample certificate to trusted directory):
[image]

Server from UA SDK client (with or without security):
[image]
Provide a certificate that the server trusts or disable certificate checks and it works. After trusting (used certificate where the host is set to my computer's name):
[image]

NikolayStek, 2020-08-24:
Hello nauful,

After a few days of test I have a results

1. Your advice for UA SDK helped thanks!

2. Foud the reason of _ActivateSessionRequest_ Decode problem (Connect to your Server).
The clients which have problem with connection to your Server send in _ActivateSessionRequest_ Nullable _ClientSoftwareCertificates_ array (-1).

If you have some time can you look at code how it can be solved in server side, because this clients Convertersystems\opc-ua-client and some proprietary software client don't have a problem with connection to UA SDK Server and UA Foundation server examples

3. Problem with _BadDissonect_ happens because Server did't answer for ClientCloseConnectionResponse

nauful, 2020-08-24:
Should be fine to skip reading ClientSoftwareCertificates in any case, I just need to get past it to read the remainder of the request. Made a fix, please update and try now.

NikolayStek, 2020-08-25:
Fix helps. Now the clients with null _ClientSoftwareCertificates_ can connect to Server.

The only _BadDissonect_ problem happens because Server did't answer for ClientCloseConnectionResponse, but it is anouther issue and not so critical.

This issue solved.

Thank's for fast response.