nauful/LibUA #33: Move common code in shared project

LadislavSopko, 2020-08-18:
Hi, I put common (Core & Framework) in shared project.

[https://github.com/LadislavSopko/LibUA/commit/edcfdd1fc54db66084c2fb5d81dc0c81e35ed660](https://github.com/LadislavSopko/LibUA/commit/edcfdd1fc54db66084c2fb5d81dc0c81e35ed660)

I let just security.cs separated, it have many differences. Maby if some one understund it better, can make it better.

Ladislav

nauful, 2020-08-18:
It is possible, but for reasons of simplicity and potential C# language changes going forward, I prefer to leave the two implementations separate. Security.cs replaces old Microsoft code with new .NET Core cryptography libraries.

closed 2020-08-18 by nauful

LadislavSopko, 2020-08-19:
I consider separate when changes happen. But its my personal opignion, I continue on my version, its too slow to mantain 2 code bases. Im adding also standardised logging, and other things, maybe check some times, something can be handy also for you.

Ladislav