nauful/LibUA #34: BadUnknownResponse during ActivateSession

alecava58, 2020-09-25:
It's the first time I use this library and I try to wrote e test program using this library but I get a BadUnknownResponse during the ActivateSession method, what did I make wrong? see the followin snippet:

Thanks in advance

            appDesc = new ApplicationDescription("urn:DyeMrp", "uri:DyeMrp", new LocalizedText("UA client"), ApplicationType.Client, null, null, null);

            client = new Client(args[0], 4840, 1000);
            var connectRes = client.Connect();
            if (connectRes != StatusCode.Good)
            {
                Console.WriteLine($"OPC OpenChannel Connect error: {connectRes}");
                Environment.Exit(1);
            }
            Console.WriteLine(connectRes);

            var openRes = client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);
            if (openRes != StatusCode.Good)
            {
                Console.WriteLine($"OPC OpenChannel OpenSecureChannel error: {openRes}");
                Environment.Exit(1);
            }
            Console.WriteLine(openRes);

            var createRes = client.CreateSession(appDesc, "urn:ProcessMachine", 120);
            if (createRes != StatusCode.Good)
            {
                Console.WriteLine($"OPC OpenChannel CreateSession error: {createRes}");
                Environment.Exit(1);
            }
            Console.WriteLine(createRes);

            var activateRes = client.ActivateSession(new UserIdentityAnonymousToken("0"), new[] { "it" });
            if (activateRes != StatusCode.Good)
            {
                Console.WriteLine($"OPC OpenChannel ActivateSession error: {activateRes}");
                Environment.Exit(1);
            }
            Console.WriteLine(activateRes);

nauful, 2020-09-27:
The code pasted should work. Could you share a network capture (Wireshark) or server that I can try running?

alecava58, 2020-09-28:
Thanks for replying.

I'm trying to write a client connecting with a machine, I suppose that the server is the machine itself (I don't have much information from the machine supplier). 
Uaexpert client is connecting successfully.
I'll try with wireshark.

nauful, 2020-09-29:
If you can provide a Wiresharp capture of the libua client connecting to the server and of uaexpert connecting to the server, I should be able to figure it out. Otherwise I need more information about this server, it seems to reject the ActivateSession request.

alecava58, 2020-10-01:
Hi nauful,

attached you can find the UAExpert capture and my test project capture. Hope they are what you need.

Thanks again for looking at my problem.

[capture.zip](https://github.com/nauful/LibUA/files/5314481/capture.zip)

nauful, 2020-10-02:
In the capture "capture my testOPC", I don't see an ActivateSession call. I see a call to Read, which the server responds to by disconnecting the client.

[image]

alecava58, 2020-10-02:
Sorry, after tenth of try I sent the wrong capture. On Monday I'll the correct one.

alecava58, 2020-10-06:
Hi, here you are the complete capture.
[capture my testOPC_activate.zip](https://github.com/nauful/LibUA/files/5334932/capture.my.testOPC_activate.zip)

nauful, 2020-10-07:
What happens if you change the locale in ActivateSession from "it" to "en-US"? This is the only difference I could find.

alecava58, 2020-10-12:
I tried with en and en-US without success. Here there are the logs.

[capture my testOPC_activate_with_en.zip](https://github.com/nauful/LibUA/files/5363627/capture.my.testOPC_activate_with_en.zip)

nauful, 2020-10-12:
Hmm... perhaps try this (different token for anonymous identity):
var activateRes = client.ActivateSession(new UserIdentityAnonymousToken("Anonymous"), new[] { "en-US" });

alecava58, 2020-10-19:
Hi, using the suggested ActivateSession with Anonymous has been succesfully!
Attached the captured file.

Next step I've tried to read and write some variables. Write had errors but was a device failure. 

[Downloads_new.zip](https://github.com/nauful/LibUA/files/5401199/Downloads_new.zip)

Thanks a lot for your valuable support.

closed 2020-10-19 by alecava58

nauful, 2020-10-19:
Try to write with UaExpert and capture that. If direct write won't work, perhaps there is a call to write similar to this: https://github.com/nauful/LibUA/issues/35#issuecomment-707167596

lortolan79, 2022-10-27:
Hello,

I have the same issue with User&Pwd.
The connection with anonymous token works fine but if I use a pwd then the ActivateSession returns "BadUnknownResponse".

[image]

Here below the Wireshark trace for ActivateSession:

[image]

And the UAExpert (that works):

[image]

I can see two differences:

"Security Sequence Number" (53 vs 54)
"Security RequestId" (3 vs 4)

But I don't know how to set these two parameters.

Thanks in advance for a reply.

nauful, 2022-10-27:
Security policy is sent as RsaPkcs15Sha:
https://github.com/nauful/LibUA/blob/bb2c24b02887be40f3c4cb16b7d193f0c1dbf0d8/NET%20Core/LibUA/Client.cs#L1730
I haven't implemented OAEP for `UserIdentityToken`, a clear password with Sign&Encrypt security is secure enough.

If you change the algorithm there, also change the `Algorithm` parameter here:
```
var activateRes = client.ActivateSession(
	new UserIdentityUsernameToken(usernamePolicyDesc, "Username",
		(new UTF8Encoding()).GetBytes("Password"), Types.SignatureAlgorithmRsa15),
	new[] { "en" });
```

lortolan79, 2022-10-28:
Dear nauful,

in my first try I used "Types.SignatureAlgorithmRsa15" as you suggest but it didn't work so in my second try I changed that parameter in OAEP encryption because UAExpert showed this type of encryption:

[image]

And UAExpert works fine.
I tried also all Types without success.
Can you help?

Thanks in advance

nauful, 2022-10-28:
You have to also implement OAEP as a crypt method here:
https://github.com/nauful/LibUA/blob/bb2c24b02887be40f3c4cb16b7d193f0c1dbf0d8/NET%20Core/LibUA/Client.cs#L1730

I haven't implemented this before so the functionality is missing. I will add it to my list of features to research later.

nauful, 2023-01-16:
Implemented more security policies and user identity token signatures (rsa-oaep) here:
https://github.com/nauful/LibUA/commit/6534b413b5abaca258542aabd81cf85094c18838