nauful/LibUA #35: Exception during call CreateSession

paolotraversi, 2020-10-07:
Hello, this is the first time I am using this library.
I wrote a test program to connect to an OPC UA server located on a remote VPN server. The server is accessible without any type of encryption or username / password.
The library goes into exception during the call to the CreateSession, in the DECODE function as the byte ENCODINGMASK is not recognized.
This is the part of the code:

      var appDesc = new ApplicationDescription("urn:Demo","uri:Demo",new LocalizedText("UA client"),ApplicationType.Client,null,null,null);

      // Create new client object to reset previous secure channel settings
      var client = new Client("172.16.34.215", 4841, 1000);
      var connectRes = client.Connect();
      if (connectRes != StatusCode.Good)
      {
        Console.WriteLine($"OPC Connect error: {connectRes}");
        Environment.Exit(1);
      }

      var openRes = client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);
      if (openRes != StatusCode.Good)
      {
        Console.WriteLine($"OPC OpenSecureChannel error: {openRes}");
        Environment.Exit(1);
      }

      var createRes = client.CreateSession(appDesc, "urn:Demo", 120);
      if (createRes != StatusCode.Good)
      {
        Console.WriteLine($"OPC CreateSession error: {createRes}");
        Environment.Exit(1);
      }

      var activateRes = client.ActivateSession(new UserIdentityAnonymousToken("0"), new[] { "en" });
      if (activateRes != StatusCode.Good)
      {
        Console.WriteLine($"OPC ActivateSession error: {activateRes}");
        Environment.Exit(1);
      }

Can I have some ideas?

Thanks,
Paolo

nauful, 2020-10-07:
Could you please upload a Wireshark capture of network traffic?

paolotraversi, 2020-10-08:
Hello nauful this is a capture of network traffic. The IP Address is 172.16.34.215 (OPC SERVER) on port 4841
Thank you for help
[image]

nauful, 2020-10-08:
Could you please upload the capture file so I can take a look?

paolotraversi, 2020-10-08:
Of course. I attach a zip file.
If you extract file there is "Test.pcapng" make from Wireshark.

Thanks alot for your help
[Test.zip](https://github.com/nauful/LibUA/files/5347662/Test.zip)

nauful, 2020-10-08:
Looks like SessionId is encoded as a GUID, this is a rare case. Most servers don't use GUID types for session ids. I'll try to add support to encode/decode NodeId as GUID type.

Would it be possible to use your server for testing?

paolotraversi, 2020-10-09:
Hello Nauful, unfortunately the OPC server (is a food device for meat cutting) is installed by a customer only reachable in VPN with authentication in two steps by email so it is not reachable.
Anyway, if you agree, I can test the modifications for you.
The times will lengthen, unfortunately I think it’s the only way to do that, if you’re willing to test the changes this way.
thanks a lot for all the support

nauful, 2020-10-09:
I will try to add support for encoding/decoding GUID NodeIds, will get back to you in a bit.

paolotraversi, 2020-10-09:
The device will be turned off during the weekend. I will be able to test the change from Monday.
Thank you very much for all

nauful, 2020-10-09:
Made a quick fix (fixed length 16 byte strings for the case of GUIDs), please let me know if it works.

paolotraversi, 2020-10-09:
Hello Nauful,
i try the fix and now the library doesen't go into exception, but i can't read the configuration node. I attach a zip file, with two images and the Wireshark dump.
In the first image you see the debug application, and in the second the remote OPC UA Server with the node.
I wrong the nodeId configuration or ReadValueId configuration?

Now the customer switch off device, i continue the test monday morning.

Thanks alot and have a nice weekend.
[test.zip](https://github.com/nauful/LibUA/files/5355756/test.zip)

nauful, 2020-10-12:
Looks like the requested encoding in your code is invalid for that server (RealValueId configuration). For DataEncoding, QualifiedName(0, null) instead of QualifiedName(2, null) should work in most cases. Could you also upload a Wireshark capture of UaExpert while reading that node?

[image]

paolotraversi, 2020-10-12:
[dump_UAExpert.zip](https://github.com/nauful/LibUA/files/5363554/dump_UAExpert.zip)

I attach a wireshark dump when UAExpert make a read request.
I try to change a QualifiedName(0, null) to QualifiedName(2, null), but the result is the same, i can't read anything.

nauful, 2020-10-12:
I think you might have uploaded an incorrect capture, this one doesn't contain any opcua traffic. The previous two did.

paolotraversi, 2020-10-12:
Sorry, I try to create the dump asap

paolotraversi, 2020-10-12:
[dump_uaexpert_1.zip](https://github.com/nauful/LibUA/files/5365259/dump_uaexpert_1.zip)

Ok now the dump is correct.

nauful, 2020-10-12:
Made a quick change to the source, please update and try again with QualifiedName(0, null).

paolotraversi, 2020-10-12:
[image]
[dump_write_UaExpert.zip](https://github.com/nauful/LibUA/files/5365671/dump_write_UaExpert.zip)
Hello Nauful,
now i can read the nodes with QualifiedName(0, null)!!!

The last problem now i can't write a parameter.
The function "Write" return "good", but at the next read the value not changed.
I Attack a picture with the part of code and the dump of UaExpert while it write a parameter.

Thanks alot for all

nauful, 2020-10-12:
I think this server accepts a call request as a program, not direct write requests:
[image]

I don't have any samples off-hand, but there should be a call request method on the client API.

paolotraversi, 2020-10-12:
Ok, i'll search the method in client API.

Thanks you for all and for your patiente

paolotraversi, 2020-10-12:
[image]

I find client API Method "Call", with a request call "sendProgram" and now i can write the program number in the device.

closed 2020-10-12 by nauful