nauful/LibUA #37: Not able to write

alecava58, 2020-10-22:
Hi, I'm able to read successfully but when I try to write I get response status 2155020288.

This is the snippet I use:

            uint[] respStatuses;

            // send the program number, for this example the program number is 32
            client.Write(new WriteValue[]
            {
                    new WriteValue(
                        new NodeId(4, "swRemRecipe1"), NodeAttribute.Value,
                        null, new DataValue(32, StatusCode.GoodClamped, DateTime.Now))
            }, out respStatuses);

            // set the status flag to 1 
            client.Write(new WriteValue[]
            {
                    new WriteValue(
                        new NodeId(4, "swRemNewRecipe1"), NodeAttribute.Value,
                        null, new DataValue(1, StatusCode.GoodClamped, DateTime.Now))
            }, out respStatuses);

            Console.WriteLine($"response status {respStatuses[0]}");

Attached there are 2 capture files, one made with UAExpert where I successfully set the variables and the second one with my TestOPC program where, after some readings, I try to set the variable with no luck. The response status is 2155020288.

Could you pls. help me.

Thanks
[Downloads.zip](https://github.com/nauful/LibUA/files/5423857/Downloads.zip)

paolotraversi, 2020-10-23:
Hi, i've the same problem.
I attached two wireshark dumps. One dump i use to write OPC Quick Client and the write work, in the second dump i used my test application and the same Node the application failed

      object value = txtValueToWrite.Text;
      var writeNodeId1 = new NodeId(2, txtObjWrite.Text.Trim());
      DataValue val = new DataValue(value, StatusCode.GoodClamped, DateTime.Now);
      var writeValues = new[]
      {
        new WriteValue(writeNodeId1, NodeAttribute.Value, null, val)
      };

      client.Write(writeValues, out var respStatuses);

respStatuses = 2155020288

Thanks

[Dump.zip](https://github.com/nauful/LibUA/files/5429376/Dump.zip)

nauful, 2020-10-26:
Seems that the servers don't support Write, which is unusual: BadWriteNotSupported = 0x80730000
I could really use one of these for testing, if either of you could share a server executable.

@alecava58: I think you have filtered your packets incorrectly, I can only see outgoing requests but not incoming responses.
@paolotraversi: What is ip address and port of your UA server? I couldn't narrow down UA traffic in the dumps.

Perhaps this applies, where the server does writes via calls and not UA write?
https://github.com/nauful/LibUA/issues/35#issuecomment-707167596

paolotraversi, 2020-10-27:
Hello Nauful, the IP address is: 127.0.0.1 and the port 49310. The UA server is installed on my PC and the IP PLC is 10.1.1.251.

Now i talk with IT Responsable to share our VPN with you

paolotraversi, 2020-10-27:
i'm thinking, instead to connect you with OPC server thorw VPN, are you agree to connect thorw TeamViewer? i give you my ID and password, on my PC there is VisualStudio, last version of library and i leave you control and you check the problem.. what do you think about this?

nauful, 2020-10-27:
That would work, please email me at my github username at gmail.com.

Before that, could you please check the dump?
[image]
Doesn't look like any UA traffic was captured. Capture a write again using UaExpert so I can see what call is used to write.

paolotraversi, 2020-10-27:
Hello nauful, today i'm in smart working and i'm not able to connect in VPN  with PLC. Tomorrow i gonna in office and i will send you the teamviewer parameters.
the dump that i sent you last time is incorrect, tomorrow i will take new dump.

thanks alot

alecava58, 2020-10-27:
@nauful I think there was not answer at all but I'm not sure, I'll try again.

I'd try to write using "calls" but it is not clear how to do (almost for me). Do you have an example? 
Thanks

nauful, 2020-10-27:
If you can capture a working write from UaExpert via Wireshark, I can take a look. Otherwise, I wonder if your server uses a program call to write instead of a normal UA write: https://github.com/nauful/LibUA/issues/35#issuecomment-707167596

This is how to make a program (server program) call with a client: https://github.com/nauful/LibUA/issues/35#issuecomment-707190499

alecava58, 2020-10-28:
> instead of a normal UA write: #35 (comment)
> (comment)

Sure, I'll do a new capture today.
Thanks

Alessandro

paolotraversi, 2020-10-28:
> If you can capture a working write from UaExpert via Wireshark, I can take a look. Otherwise, I wonder if your server uses a program call to write instead of a normal UA write: [#35 [...]
I attach new dump, the picture show wireshark filter find tcp:port==49310.
Today if you want to use TeamViewer i will send you the parameter

[Dump_write_opcQuickClient.zip](https://github.com/nauful/LibUA/files/5450506/Dump_write_opcQuickClient.zip)
[image]

nauful, 2020-10-28:
It still doesn't look like there is any UA traffic in your capture. Can you verify that the opcua filter in Wireshark is able to recognize UA packets?

paolotraversi, 2020-10-28:
This is very strange. Wireshark configuration is the same of the first dump that i send you for my first problem. The OPC server are different.

[image]

Now i change the port for OPCUA (see the picture). Before the port was 4840, now 49310.
[Write_QuickOpc.zip](https://github.com/nauful/LibUA/files/5451462/Write_QuickOpc.zip)

I Attack you a new dump

alecava58, 2020-10-28:
@nauful here you are the new write capture using UAExpert and my test program capture using server calls. The server 
call seems to be succesfull, no error returned, but the variable has not changed. 

[capture2810.zip](https://github.com/nauful/LibUA/files/5451513/capture2810.zip)

nauful, 2020-10-28:
@alecava58 Did the write from UAExpert succeed? I don't see any responses in the capture:
[image]

If the variable didn't change and there was no error, writes for that variable probably don't work on the server side.

alecava58, 2020-10-28:
Yes, the writes using UAExpert succeed. Next time I'll send the screenshot of the ui.

alecava58, 2020-10-28:
@nauful wait wait  , reviewing the capture file with wireshark I realized that may be I used a wrong filter.
Let me redo the capture session and check.
Sorry

alecava58, 2020-10-29:
@nauful here you are the new capture files NOT filtered during capture. The IP address of the UA server is 192.168.78.29 so you should use this address as filter during view.
There is a capture using UAExpert to read and write the variables and my testOPC program with read and 2 write methods; directly and using calls program.
Hope this helps you to understand what is the problem.

Many thanks in advance
[capture2910.zip](https://github.com/nauful/LibUA/files/5457684/capture2910.zip)

nauful, 2020-10-29:
Looking at the response in testOPC:
[image]

Try changing this:
```
        client.Write(new WriteValue[]
        {
                new WriteValue(
                    new NodeId(4, "swRemRecipe1"), NodeAttribute.Value,
                    null, new DataValue(32, StatusCode.GoodClamped, DateTime.Now))
        }, out respStatuses);
```

To this:
```
        client.Write(new WriteValue[]
        {
                new WriteValue(
                    new NodeId(4, "swRemRecipe1"), NodeAttribute.Value,
                    null, new DataValue(32))
        }, out respStatuses);
```

And the same for the other node.

paolotraversi, 2020-10-29:
Hello Nauful, i try to change my software, but the result is the same.

alecava58, 2020-10-29:
@nauful good news from my side. Using your latest tips I was able to write the variables.
Attached you can find the capture of my test program where you can see (I think) the succesfully writes.
Thanks again for your help.

[capture ok 2910.zip](https://github.com/nauful/LibUA/files/5459225/capture.ok.2910.zip)

paolotraversi, 2020-10-30:
@nauful i try to use another paid library (we want to stop use it) and i use your trick to remove parameters STATUS CODE and TIMESTAMP in write method (in paid library) and now i was able to write.
I try to  use same trick in your library but the response are always 2155020288 BadWriteNotSupported

Any Ideas?

nauful, 2020-10-30:
Please upload a capture of the write that works and the write that doesn't.

paolotraversi, 2020-10-30:
@nauful i attach two dumps
[Desktop.zip](https://github.com/nauful/LibUA/files/5465250/Desktop.zip)

nauful, 2020-10-30:
I can't find UA traffic in your dump. Could you capture and filter by opcua?

paolotraversi, 2020-10-30:
@nauful i use a remote server OPCUA e not local. I verified dumps and i found filter opcua OPCUA and TCP.PORT = 49310
[Desktop.zip](https://github.com/nauful/LibUA/files/5466160/Desktop.zip)

nauful, 2020-10-30:
The one that works is writing int16 (short):
[image]

The one that fails is writing int32 (int):
[image]

Try writing using a short.

nauful, 2020-11-02:
@paolotraversi Let me know if writing a short works, or if you'd like me to investigate via TeamViewer.

paolotraversi, 2020-11-03:
@nauful sorry but yesterday i wasn't at work.
how can i write using short if the "DataValue" use "Object" to define the value? i try to using cast, but the result is the same...

nauful, 2020-11-03:
Try this?
```
Int16 v = 314;
client.Write(new WriteValue[]
{
    new WriteValue(new NodeId(4, "swRemRecipe1"), NodeAttribute.Value, null, new DataValue(v))
}, out respStatuses);
```

Let me know if that doesn't work, then I'll try TeamViewer.

paolotraversi, 2020-11-03:
@nauful this code doesn't work for me. If you can try to investigate via teamviewer now i go to lunch, i come back to office from one hour. i will send you email when i gonna be in office with teamviewer password.

closed 2020-11-03 by nauful