nauful/LibUA #39: Need help, please

d1den, 2021-02-08:
I'm researching your library and wanted to implement parent node creation and deletion. To delete, I wrote the following method:  
`public void TryRemoveChieldNode(Node chieldNode, Application application)
			{
				try
				{
					this.References.Remove(this.References.Find(refNode => refNode.Target == chieldNode.Id));
					chieldNode.References.Remove(chieldNode.References.Find(refNode => refNode.Target == this.Id));
					Node node;
					application.AddressSpaceTable.TryRemove(chieldNode.Id, out node);
				}

                catch { }
			} `  
The node becomes unavailable but is not deleted on the client. Tell me, if possible, how to remove a node? I think it is necessary for the client to call HandleBrowseRequest (). How to do this, can you tell me, please?

d1den, 2021-02-08:
Also just now found that adding a node on the server also doesn't show up in the client. After reconnecting the client, all added nodes are displayed correctly. It turns out that you need to update the display of nodes in the client to dynamically add and remove nodes. But I don't understand how to do this :(. Hope for your help, thanks in advance.

nauful, 2021-02-10:
Hello,

Quick overview of nodes in UA, if you want A as a parent of B:
Node A is created. Node B is created.
Add to A's references: B is child.
Add to B's references: A is parent.

AddressSpaceTable is a quick dictionary that lets you look up node IDs to nodes.

From a client's perspective, this is how node discovery works:
Browse(A id) -> gets a reference (B id, child)
Read/Browse/next operation for (B id)

To remove a node on the server:
Remove all references to this node from other nodes (usually parent/children in a hierarchy if no other fancy relationships exist).
Remove from AddressSpaceTable.

Clients need notifications when the model is changed (i.e. node added/removed), which they won't figure out unless browse is called. To handle this, I recommend publishing a model changed notification from the server node (ns=0;i=2257, it's a constant in the library and in OpcUA), and subscribing to that from your client. If it's your own client, I could give you some suggestions on how to do that. If it's not your own client, it would just get errors on Read or missing nodes on Browse and adjust itself.

d1den, 2021-02-10:
The client is not my own. Can you please tell me how to publish a change notification on the server? I don't understand how to do this

nauful, 2021-02-10:
If it's not your own client, adding change notifications won't help because it's not standard behaviour. Could you give me more information about the client? The issue here is with the client, not the server.

d1den, 2021-02-10:
I used [Prosys OPC UA Client](Download "https://drive.google.com/file/d/1GFJm2dYCUaUfSnczAOrfNfNmDEmWrss9/view?usp=sharing").

nauful, 2021-02-10:
If you use that as a library, are you able to subscribe to arbitrary nodes?

d1den, 2021-02-22:
No, I used this as a test client.  Since this is not the standard behavior as you said, I decided to leave the standard behavior.  Thank you for your help!

closed 2021-02-22 by d1den