nauful/LibUA #41: Fehlende Item-Ändeurngen

seti247, 2021-03-08:
Hallo,

ich versuche einige Items zu überwachen. 
Generell funktioniert es, allerdings werden dabei immer sporadisch Wertänderungen übersprungen und die Methode NotifyDataChangeNotifications nicht ausgeführt.
Die Verbindung bleibt aber bestehen. Danach geht es dann wieder.

Habe ich bei der Initalisierung etwas falsch gemacht oder hätten Sie eine Idee wo das Problem liegen könnte?

Vielen Dank schon mal :-)

```
Public Sub Create()
            _client = New OPCClient(SERVER_ADDRESS, SERVER_PORT, 1000)
            _client.Connect()

            If _client.IsConnected Then
                Dim applicationDescription As New ApplicationDescription(
                        "urn:" & APPLICATION_NAME,
                        "uri:" & APPLICATION_NAME,
                        New LocalizedText(APPLICATION_NAME),
                        ApplicationType.Client,
                        Nothing,
                        Nothing,
                        Nothing
                    )

                _client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, Nothing)
                _client.CreateSession(applicationDescription, "urn:" & APPLICATION_NAME, 120)
                _client.ActivateSession(New UserIdentityAnonymousToken("0"), {"en"})

                _client.CreateSubscription(0, 1000, True, 0, _subscrId)

                Dim respStatuses As UInteger() = Nothing
                _client.SetPublishingMode(True, {_subscrId}, respStatuses)

                Dim monitorCreateResults As MonitoredItemCreateResult() = Nothing
                _client.CreateMonitoredItems(
                    _subscrId,
                    TimestampsToReturn.Both,
                    New MonitoredItemCreateRequest() {
                        New MonitoredItemCreateRequest(
                            New ReadValueId(
                                New NodeId(3, NODE_NAME),
                                NodeAttribute.Value,
                                Nothing,
                                New QualifiedName()
                            ),
                            MonitoringMode.Reporting,
                            New MonitoringParameters(0, 0, Nothing, 100, False)
                        )
                    },
                    monitorCreateResults
                )
            End If
End Sub
```

```
Imports LibUA
Imports LibUA.Core

Public Class OPCClient
    Inherits Client

    Public Sub New(ByVal Target As String, ByVal Port As Integer, ByVal Timeout As Integer)
        MyBase.New(Target, Port, Timeout)
    End Sub

    Public Overrides Sub NotifyDataChangeNotifications(subscriptionId As UInteger, clientHandles As UInteger(), notifications As DataValue())
        For i = 0 To clientHandles.Length - 1
            Log(String.Format("subscrId {0} handle {1}: {2}", subscriptionId, clientHandles(i), notifications(i).Value.ToString()))
        Next
    End Sub
End Class
```

nauful, 2021-03-09:
Client looks fine, it's a server issue.

I suggest trying CreateSubscription with RequestedPublishingInterval > 0 (i.e. 10 ms), it's an issue for timer-driven publishing with servers. Also try reducing MaxNotificationsPerPublish to maybe 64 and go up from there if it's stable with no discarded values.

seti247, 2021-03-12:
Hi,

the first impression looks very good. 
It seems to be working now. 

Many Thanks!

closed 2021-03-12 by seti247