nauful/LibUA #43: Server: Monitor value change

d1den, 2021-03-10:
Hi!
I need to monitor a node on a client. But there is no value displayed in the client monitoring window. 
[image]

How to set up monitoring so that new values are displayed correctly?
Need this function to display updated values? How to use it (I didn't succeed)?
[image]

Thank you in advance

nauful, 2021-03-10:
If you subscribe to a node in the sample server application from the client, do you see new values getting published?

d1den, 2021-03-10:
No, it is displayed as in the screenshot above (DataValue is not displayed)

nauful, 2021-03-10:
What happens if you use UaExpert instead of the client?

Could you also take a Wireshark capture of network traffic?

d1den, 2021-03-11:
Tried monitoring in UaExpert:  
[image]

nauful, 2021-03-11:
In that version of UaExpert, what does it look like if you subscribe to a publishing node in the sample test server?

Also, could you take a Wireshark capture of network traffic with security set to None?

d1den, 2021-03-15:
Several tests:  
1. Sample test server monitoring in UaExpert:  
[image]
It`s working, but some error.
2. In Matricon Opc Ua Explorer:  
[image]
When you add a variable to monitoring, an error is generated and the connection is disabled.  
3. In Prosys Opc Ua Client:  
[image]

And I will do Wireshark Capture soon.

d1den, 2021-03-15:
Wireshark Capture in multiple programs: https://drive.google.com/drive/folders/1t0SpjmXAlYrrWpVx9pHYESY1yooEb1R-?usp=sharing

nauful, 2021-03-16:
Can you upload the 3 clients?

d1den, 2021-03-16:
The clients: https://drive.google.com/drive/folders/1p7z_6tT8pFVGvst7kMaDOnXu_RhMpVnB?usp=sharing

nauful, 2021-03-16:
For uaexpert, looks like PublishRequest includes an array of acknowledged publishes (subscription id, sequence number) and expects the next PublishResponse to include a list of status code updates on those acknowledges if I understand correctly. Those errors show up sporadically due to timing differences between uaexpert acknowledging a publish and the next response containing a useless "good" for that.

Not sure how fixing that would work because publishes are done asynchronously for performance reasons (either on timer, immediately when queued or triggers), but it's safe to disregard that warning I believe.

Matrikon is calling SetMonitoringMode, which isn't currently supported. I'll add support for that.

Looks like Prosys isn't sending any publish requests, will look for a workaround.

nauful, 2021-03-17:
Matrikon fixed: https://github.com/nauful/LibUA/commit/4b6df3f06ce5f46ef57c1ccda01444491b055c75

nauful, 2021-03-19:
I've added more UA address space variables that Prosys UA client tries to read when connecting, but it still won't send a PublishRequest.

Previous issue here: https://github.com/nauful/LibUA/issues/25

Closing for now, until I find more information about Prosys client. Other two clients are working now.

closed 2021-03-19 by nauful

d1den, 2021-03-19:
Thank you for the help!