nauful/LibUA #47: Handle OnSubscribe/OnUnsubscribe

IvoFortunov, 2021-03-26:
Hello, I'm trying to use your library in one of my projects.
I want to handle OnSubscribe and OnUnsubscribe events.
The idea is to find which nodes are subscribed in a given moment and provide values for them. In my project values are read asynchronously, so I need to tell the other part of the program to read from the external device.

nauful, 2021-03-26:
Under Application:
https://raw.githubusercontent.com/nauful/LibUA/75b36c4894a8bd07c2d13d9078d9e5765890cee1/NET%20Core/LibUA/Application.cs

You can override these two. Remember to call the original methods after overriding. 
public virtual bool MonitorAdd(object session, MonitoredItem mi)
public virtual void MonitorRemove(object session, MonitoredItem mi)

MonitorAdd is called whenever a node id is monitored (monitored item request add to subscription), MonitorRemove is called whenever a node id is unmonitored (monitored item request remove from subscription or subscription closed). Note that MonitorAdd and MonitorRemove might be called multiple times for the same node id depending on clients.

To get the current set of monitored items:
```
try
{
	monitorMapRW.EnterReadLock();
	... look at monitorMap
}
finally
{
	monitorMapRW.ExitReadLock();
}
```

IvoFortunov, 2021-03-26:
Thanks a lot!

closed 2021-03-26 by IvoFortunov