nauful/LibUA #48: Cannot connect to SIMATIC WinCC UA

IvoFortunov, 2021-03-29:
Hello,
I continue exploring your library, and it looks very promissing.
The final goal of my project is to connect to SIMATIC WinCC. My application is a server. Everything works fine with clients like UAExpert and other, but when I try to connect from SIMATIC WinCC I've got the following:
1. With security set to none, SIMATIC sends  "CloseSecureChannelRequest"
Wireshark log: [SecurityNone.zip](https://github.com/nauful/LibUA/files/6219516/SecurityNone.zip)
2.With security Basic128Rsa15 i get BadInternalError
WIreshark log: 
[SecurityBasic128Rsa15.zip](https://github.com/nauful/LibUA/files/6219537/SecurityBasic128Rsa15.zip)

Please help me solve this issue!
Thanks in advance

nauful, 2021-03-29:
I can't easily find UA traffic in your dump, could you filter them to only include the ports used for UA traffic?

IvoFortunov, 2021-03-29:
I think I've managed to filter them. If you write "opcua" in filter, they are filtered by protocol.

3. With security Basic128Rsa15 and the original certificate i get Unknown error, but things look better
WIreshark log:
[SecurityBasic128Rsa15_org_cer_opcua.zip](https://github.com/nauful/LibUA/files/6222784/SecurityBasic128Rsa15_org_cer_opcua.zip)

nauful, 2021-03-31:
Seems to expect a specific structure for the read value of Server_ServerStatus_State.

Could you capture traffic from a server-client configuration that works, with None security mode? That same value blob can be reused here.

IvoFortunov, 2021-03-31:
Here is working log with Unified Automation Test UA Server and last version of Simatic WinCC
Wireshark log:
[WorkingConnection_new_WinCC.zip](https://github.com/nauful/LibUA/files/6236850/WorkingConnection_new_WinCC.zip)

IvoFortunov, 2021-04-01:
Ok I fixed the connection issues. WinCC asks for "ServerCapabilities_OperationLimits" so I've added them to "internalAddressSpaceValues". It also expects UInt32 as an answer.
Now i connects and browses "Server Node" but it doesn't wrowse "Items Node".
Can you please hel me with this issue.
Wireshark log:
[Connected_Cannot_Browse_Items.zip](https://github.com/nauful/LibUA/files/6241241/Connected_Cannot_Browse_Items.zip)

nauful, 2021-04-01:
What is the value for MaxNodesPerBrowse? It seems to pick that as 0 and not browse.

IvoFortunov, 2021-04-01:
No it is 100, and yet another strange thing - it seams to ask for NodeId(0,0) instead of (2,0) when I try to browse Items folder.
It looks like that WinCC thinks that Items have nodeId=0
[image]

In the working configuration it seams to ask for the correct NodeId.

nauful, 2021-04-01:
Could you make a pull request to merge your fixes into the main library?

From the root, there should be a link to ns=2;i=0 that represents items. It seems that the namespace is dropped? I'm not sure how to fix that.

IvoFortunov, 2021-04-02:
I continue investigating, when I finish I will make a pull request with all the fixes.

IvoFortunov, 2021-04-02:
It is very strange behaviour. When I've changed NodeId(2,0) for Items to NodeId(2,"Items") - string identifier WinCC started to browse the Items.

nauful, 2021-04-02:
Is the browse link from root to items the same i.e. ns=2;i=0 and also updated similarly to ns=2;s="Items"?

I wouldn't use this change for other servers, many (especially embedded) don't allow string identifiers.

IvoFortunov, 2021-04-02:
No this change is only for my project!
I've just changed ns=2;i=0 to ns=2;s="Items".
For now we only have to add ServerCapabilities_OperationLimits" (12 values) to "internalAddressSpaceValues".
Now I have other issues, and when I solve them we can decide together what to include in the library.

P.S. Can we communicate by email, when there are things to discuss?

nauful, 2021-04-02:
You can email me: github username @gmail

IvoFortunov, 2021-04-03:
It finally works:
[image]
Thanks for the help!

closed 2021-04-03 by IvoFortunov