nauful/LibUA #51: Status of pub/sub in LibUA server?

PrecoSWEng99, 2021-04-23:
I'm using your open source LibUA package to experiment with your OPC UA Server.  I downloaded a copy back in January 2020 and haven't updated it since.  We have a couple of machine control programs that currently contain DA servers, and we want to revise them to use UA.  So I've put your LibUA server code into a class library for testing.  So far we are pleased with the results.  I typically use the UaExpert client for connections to the server.  It looks like the publish/subscribe features are not finished yet.  In UaExpert I can drag a tag from the Address Space panel into the Data Access View panel.  That causes the client to request that the server create a new subscription via message type 787 (and the server does create the subscription).  Then the client requests that the tag be placed in a monitored item list via message type 751 (looks like that is also working).  But when the client sends a publish request message (type 826) to the server, nothing else happens.  Do you have any information or advice about this 
problem?

I see a function MonitorNotifyDataChange() that is never called.  Was that intended to be part of the pub/sub scheme?

nauful, 2021-04-24:
Pub/sub has been working since the initial release. Try UaExpert with the latest sample server and subscribe to any Trend XYZ item under Items.

You need to call MonitorNotifyDataChange from your application to publish values. This line shows how the sample server enqueues change notifications:
https://github.com/nauful/LibUA/blob/fbbdf2bb0cb6c46211266d9ffe5b1ade635ece61/NET%20Core/TestServer/Program.cs#L422

If you still don't get any publishes, in Application.cs, check that MonitorAdd is getting called when the client is connected, and that monitorMap has monited items added.

nauful, 2021-04-29:
Tested with uaexpert and sample server, subscriptions and publishes are working.

closed 2021-04-29 by nauful