nauful/LibUA #54: Problems creating methods with input and output arguments

jpramos1993, 2021-05-21:
This might seem like a newbie question but i cant seem to figure out how to create methods with input and output arguments.

I created a server with several folders, properties and methods, and all are being displayed correctly in UaMaster. However all the methods created do not have input arguments and/or output arguments(the node exists but not the argument value array).

I already tried to add arguments using references, fiddled around with the input arguments value and with the HandleReadRequestInternal to no avail.

Can i get some tips?

Thanks in advance.

jpramos1993, 2021-05-21:
i was seeing the other tickets and from what i understand this implementation does not support the definition of methods that appear in UaMaster with input args or output args. 
However the Call method can be performed with a random/specific number of input parameters from the client(which is awesome by the way) but i am still having problems with the encoding of the result to send back to the client

nauful, 2021-05-23:
On the server side: https://github.com/nauful/LibUA/blob/9a3ef5c1ecc691914ea2c7e6b7b2e0d4712074bc/NET%20Core/LibUA/NetDispatcher.cs#L2770

Before line 2806, you might add something along the lines of
```
StatusCode callResult = app.HandleCall(reqs[i]);
succeeded &= respBuf.Encode(callResult);
```

You would also need to add a HandleCall method to Application (default method might just be `return StatusCode.BadMethodInvalid`) and override that in your own application. I didn't do very much on this because the standard for method calls varies based on client.

jpramos1993, 2021-05-23:
Thanks a lot for the help on this topic and for providing the community with this amazing package.

I guess am a newbie with opc ua and was expecting a more restrictive communication between server and client, but i kinda prefer this more agile approach for applications.

closed 2021-05-23 by jpramos1993

nauful, 2021-05-23:
UA mostly just defines a communication specification. There are some models for address space in the specification, but these are loosely compatible between vendors.