nauful/LibUA #58: Connection problem with Prosys Client v4.0

Polyteres, 2021-07-06:
Hi, we are having some issues with the Test Server and the Prosys OPC UA Browser. With a older version of Prosys Client (version 3.0), we had not any problems, but with the new version (4.0) the server refuse de connection with a certificate error, and we do not get any information about what is happening.

The client can be donwload from here https://www.prosysopc.com/products/opc-ua-browser/ 

Thank you very much for your support.

nauful, 2021-07-07:
There isn't a direct download link on the page. Please upload the client to dropbox or another host where I can download it directly.

Polyteres, 2021-07-09:
Hi nauful, you can download from here https://www.prosysopc.com/opcua/apps/UaBrowser/dist/4.0.6-268/prosys-opc-ua-browser-windows-x64-4.0.6-268.exe

Thank you.

nauful, 2021-07-11:
[image]

It seems the client does not like a browse response, then closes its connection. I do not see a certificate error. Could you upload a capture with Wireshark?

nauful, 2021-07-15:
This is beyond the scope of a simple bug fix on the server side because I don't see any error or anything wrong.

Most likely there is something the client is looking for but doesn't specify, and finding that would take significant reverse engineering effort, which is more than I have time for at the moment. If your company is working on a commercial project and would like my help in this area, I could do it as a consulting arrangement. Feel free to contact me: github username @gmail.com

closed 2021-07-15 by nauful

gogolevsergey, 2023-06-23:
I investigate this problem and found, that Prosys OPC UA Browser want that ShutdownReason internal variable will be LocalizedText, otherwise it still requesting this nodes and don't get to browse requests. To fix this you need add these internal variables at SetupInternalAddressSpace method of Application.cs. I will propose PR soon.
[image]

nauful, 2023-06-23:
Thanks, looking forward to it!

gogolevsergey, 2023-06-23:
Problem seems to be more complicated, will investigate more, seems this client send some strange browse request and then disconnects, after this establish session with repeatedly requests to read ServerStatus and BuildInfo nodeids. And finally sometimes it reads address space, very strange...

nauful, 2023-06-23:
It may be trying and failing to subscribe to some particular node.

franklupo, 2023-07-21:
Hi,
I use version 4.4.0 and I get an error while connecting

```System.Net.Sockets.SocketException: 'Connessione in corso interrotta forzatamente dall'host remoto.'```

best regards

nauful, 2023-07-21:
That doesn't look like a library issue. I would guess a firewall is terminating your connection.