nauful/LibUA #60: Help with Nodemethods and arguments

madagaga, 2021-07-22:
Using Ua expert as client, I've successfully created the node tree with a node method and a nodemethod with argument.
But I can't browse correctly the InputArguement node. 
I've noticed that ExpandedObject is not really used, I already added it in "GetVariantTypeFromType" 

I think the Argument object is missing, and I don't know how to encode it correctly.

nauful, 2021-07-22:
Please capture the traffic with Wireshark and upload it here. I can take a look and tell you what is missing. Most likely you'd have to serialize into an ExtensionObject. I don't have any samples of doing this and the UA specification is unclear.

madagaga, 2021-07-23:
Yes, i saw in other projects, Argument object is set into a ExtensionObject. 
The extension object serialization had an issue. Now it's OK. 
Last thing is then encoding mask : 

In wireshark expected value is : "EncodingMask: 0x01, has binary body" 
What i have is : "EncodingMask: 0x01, EncodingMask: Four byte encoded Numeric" 

In ua expert, tree values are correct, but when i call the method, ua plugin says that method has no arguments 

Sorry if i'm unclear

madagaga, 2021-07-23:
Yes, I checked on other projects. 
So I created a Argument object and encoded it correctly in an ExtensionObject. 
My mistake was on the string encoder.
Now it's OK on ua expert but only in the tree view, I can see all nodes and arguments. 
The last thing is to be able to call the method with arguments in UA expert. 
I continue to check. 
Thanks for the help !

madagaga, 2021-07-23:
Ok found. 
There is an issue in HandleTranslateBrowsePathRequest 
Line `if (target.BrowseName.Equals(rp.TargetName))`
should be 
`if (target.BrowseName.Name.Equals(rp.TargetName.Name))`
as ua expert doesn't send the good namespace identifier in the request .

closed 2021-07-23 by madagaga

nauful, 2021-07-23:
Just took a look at the source, both BrowseName and TargetName are QualifiedNames. This means that equality includes a test for namespace and value. This is likely a bug in uaexpert, namespaces are independent and comparing values across namespaces shouldn't match.