nauful/LibUA #61: wrong return value, {(null)} instead of null, of the IsAbstract attribute of the TestServer

Ttberg, 2021-08-02:
Around last year I tried to create an own client application, and eventually succeeded. Now I wanted to try and create an own server application. This seemed to work at first (I could connect with the opc reference client to the TestServer and read some values of the TestServer, where I used the version of Apr 21, 2020 of LibUA), but after updating the LibUA to the latest version it failed, the opc reference client crashed with a null reference, because it tried to read the IsAbstract attribute, but the TestServer returns {(null)}.

There has been a change in the Application.cs, in the function HandleReadRequest(...), where eventually a new DataValue(null, StatusCode.BadAttributeIdInvalid) is returned in the old version, but in the latest version a new DataValue(null, StatusCode.Good) is returned. If I use the latest LibUA version but use StatusCode.BadAttributeIdInvalid, the opc reference client does not crash, and I can read the TestServer values again.

However, after setting a breakpoint in the opc reference client code, I could see the TestServer returns a null value in the old LibUA version, but in the new version it returns a {(null)} value, and therefore the reference client crashes. So I don't think that the  change of the StatusCode is the issue, but that some other change in LibUA causes this issue?

nauful, 2021-08-02:
Please upload a binary executable of the test client. I will see what I can sniff with Wireshark and go from there.

Ttberg, 2021-08-02:
[ReferenceClient.zip](https://github.com/nauful/LibUA/files/6918234/ReferenceClient.zip)
I attached a zip file containing all dll's and the Quickstarts.ReferenceClient.exe

please note that the exception is not visible in this case, it is only visible when debugging with visual studio. In that case, a null exception is thrown, when stepping through, the data of the server will still be visualized. When you run the executable, you will not be able to see the exception, but you should of course still be able to see the difference in the return values using Wireshark.

nauful, 2021-08-02:
Accidently missed this about 3 weeks back. Fixed and pushed, please try now.

Ttberg, 2021-08-03:
yes, this does fixes it, thanks!

I was a bit confused about the {(null}) value, but this value is actually discarded by the reference client if the server returns a BadAttributeIdInvalid, so everything gets handled correctly afterwards.

closed 2021-08-03 by Ttberg