nauful/LibUA #64: Unable to connect if client sends MaxMessageSize=0 in Hello

heppth, 2021-08-11:
Hello,

First of all, compliments for the library. It is fun to work with.
I have detected a small issue. My OPC-UA client can not connect. The server has following output:

> [Error] Sent TL error 0x80080000

My client sends MaxMessageSize=0, which indicates that the Client has no limit. The LibUA server then creates a MemoryBuffer with size 0.

Setting MaxMessageSize to zero is valid. See [specification](https://reference.opcfoundation.org/v104/Core/docs/Part6/7.1.2/)

amsico, 2021-08-11:
I had a similar problem when using another library which uses [MaxMessageSize=0](https://github.com/FreeOpcUa/python-opcua/blob/350ef18e399ac6ce80f987992ee06e0c9e3b1ca3/opcua/client/client.py#L138) as described above.

To solve the issue, I modified the [Client.cs](https://github.com/nauful/LibUA/blob/31c8f5ba68f9f925d93c2da1610b438f585fe0a6/NET%20Core/LibUA/Client.cs#L980) in line 980 by adding:

```
[...]
if (!recvHandler.RecvBuf.Decode(out config.TL.RemoteConfig.MaxChunkCount)) { return StatusCode.BadDecodingError; }

if (config.TL.RemoteConfig.MaxMessageSize > 0)
    MaximumMessageSize = (int)Math.Min(config.TL.RemoteConfig.MaxMessageSize, MaximumMessageSize);
```

nauful, 2021-08-14:
Thanks, fixed.

closed 2021-08-14 by nauful

krosthw, 2024-10-14:
Hello, 
has been this _if_ implemented? I don't see it in Client.cs.
I have the same problem of @amsico.