nauful/LibUA #68: System.Exception on CreateSession

loliparra80, 2021-08-23:
I have a simple test console application from your samples:

`var appDesc = new ApplicationDescription(
				"urn:DemoApplication", "uri:DemoApplication", new LocalizedText("UA SDK client"),
				ApplicationType.Client, null, null, null);
ApplicationDescription[] appDescs = null;
EndpointDescription[] endpointDescs = null;

Client clientUA = new Client("127.0.0.1", 49320, 1000);
clientUA.Connect();
clientUA.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);
clientUA.FindServers(out appDescs, new[] { "en" });
clientUA.GetEndpoints(out endpointDescs, new[] { "en" });
var createRes = clientUA.CreateSession(appDesc, "urn:DemoApplication", 120);
clientUA.Disconnect();
`

But when try to make de CreateSession I had an Exception:

[image]

This is the server configuration, I had configured without security.
[image]

If I set the security with Basic256Sha256, then a get another error:

[image]

The code with security:

`		var usernamePolicyDesc = endpointDescs
			.First(e => e.UserIdentityTokens.Any(t => t.TokenType == UserTokenType.UserName))
			.UserIdentityTokens.First(t => t.TokenType == UserTokenType.UserName)
			.PolicyId;

		clientUA = new Client("127.0.0.1", 49320, 1000);
		var connectRes = clientUA.Connect();
		clientUA.OpenSecureChannel(MessageSecurityMode.SignAndEncrypt, SecurityPolicy.Basic256Sha256, serverCert);
`

Maybe you can understand better the error than me and suggest me another configuration o whatever.

Thank you.

nauful, 2021-08-24:
Could you share the server with me so I can test this locally?

loliparra80, 2021-08-24:
Hello, today I have tried my test application directly as a new project in your LibUA solution, with the LibUA reference directly from your LibUA project, and also ser the Token with "Anonymous" and now it's working.

var activateRes = client.ActivateSession(new UserIdentityAnonymousToken("Anonymous"), new[] { "en" });

Maybe the latest release in nuget package installer is not already updated, I don't know. Thank you.

closed 2021-08-24 by loliparra80