nauful/LibUA #70: Adding AnalogItems to server setup

PaulKjetil, 2021-12-15:
The LibUA seems to be a nice solution. But I miss some docs or examples on how to build more complex objects.

Working to set up a server that has AnalogItemType as in UA Data Access or alternatively adding single EngineeringUnit to the nodevariable. 
Seems to get a problem to make a proper instance of the EngineeringUnit variable. When browsing from a client it produced nodes with the Value = null.  Is there any examples on the correct method to define the structures ?

rgds,
Paul Kjetil Andreassen

Some initial code:

         Dictionary<int, Node> MySST = new Dictionary<int, Node>();
            foreach (var tag in taglist)
            {
                if (tag.Value.OpcItemId >= 1000000)
                {
                   ntag = new NodeVariable(new NodeId(2, (uint)tag.Value.OpcItemId), new QualifiedName(tag.Value.VisibleName),
                        new LocalizedText(tag.Value.VisibleName), new LocalizedText(tag.Value.Description), 0, 0, 0,
                        AccessLevel.CurrentRead, 0, true, nodeTypeFloat);

                    mtag = new NodeVariable(new NodeId(2,(tag.Value.OpcItemId.ToString() + ".EngineeringUnits")), new 
                                 QualifiedName("EngineeringUnits"), new LocalizedText("EngineeringUnits"), 
                                 new LocalizedText(tag.Value.Name), 0, 0, 0, 0, 0, true, new NodeId(UAConst.EUInformation));

                    AddressSpaceTable.TryAdd(ntag.Id, ntag);
                    AddressSpaceTable.TryAdd(mtag.Id, mtag);

                    ntag.References.Add(new ReferenceNode(new NodeId(UAConst.HasProperty), mtag.Id, false));

                    SSTLoop.References.Add(new ReferenceNode(new NodeId(UAConst.Organizes), ntag.Id, false));
                    ntag.References.Add(new ReferenceNode(new NodeId(UAConst.Organizes), SSTLoop.Id, true));
                    MySST.TryAdd((int) ntag.Id.NumericIdentifier, ntag);

                }
            }

nauful, 2021-12-18:
UA structures are defined as nodes existing in the address space (AddressSpaceTable) linked by references, and it seems that you have both correct.

Try changing the child node of the reference from .Organizes to .OrganizedBy (relationship parent organizes child organizes parent and repeat might throw a property reading loop).

If you still see Value = null, please capture network traffic using Wireshark with no UA security. I'd like to see what 
`public virtual DataValue[] Application.HandleReadRequest(object session, ReadValueId[] readValueIds)` is receiving. If `readValueIds` target the correct node id and property, you should be getting good values back.

nauful, 2022-04-27:
That should have solved your issue. Please leave a comment if you still have this problem.