nauful/LibUA #71: Writing tag problem

byrialsen, 2022-01-24:
Hi

I assume this is more a noob question. I get this null reference exception. Do you have any idea what I do wrong. I believe it is related to the NodeId constructor.

[image]

nauful, 2022-01-25:
Hello,

You should define that NodeId as NodeId(3, "\"dbMsi\".\"toSf\".\"WatchDog\").
ns=3;s=... is the serialized form of NodeId. Alternative, you could have used NodeId.Parse with that string. Otherwise the node id will not match as it is right now.

Could you tell me what object is causing that null reference exception? It should be fine.

nauful, 2022-04-27:
That should have solved your issue. Please leave a comment if you still have this problem.

closed 2022-04-27 by nauful