nauful/LibUA #73: Setting up default address space takes a lot of memory

pspeybro, 2022-01-28:
I was trying to run the test server and noticed that during startup, the application briefly takes around 270MB of memory.
This seems to happen while setting up the DefaultAddressSpace in Application.cs when entering SetupDefaultAddressSpace()

I tried to quickly debug to see what is taking so much memory but could not identify it. It seems to spike and drop down again to acceptable levels.

[image]

This can result in Out of memory errors when running on small devices like a Raspberry PI zero W 2.
Initially I thought it might be the demo historical data, but that is not the case.

Can someone explain what is taking this memory?
Is there a way to make this more efficient (less memory use)?

nauful, 2022-01-31:
There are many small intermediate objects created while setting up the default address space. After a threshold (either time passes or memory limit reached), discarded objects are marked and feed.

This memory spike is due to how .NET's garbage collector works. When there is memory pressure, the top of the spike is clipped to not exceed the end result by more than a small margin (discarded objects are marked and collected when that happens).

For low-memory devices I wouldn't recommend .NET (I have a similar unreleased SDK for C++ which uses fixed memory allocations). If you must, I recommend not using the usual default address space and only create required nodes.

pspeybro, 2022-02-01:
I'm aware of the garbage collector, but actually, this spike already seems to happen the moment you enter the SetupDefaultAddressSpace method in the Application.cs. Even before all the items are being created. 

With .Net Framework the spike seems to be lower (83MB) but it is still weird that it seems to increase while just stepping into the method.

.Net seems to be optimizing and already allocate stuff in advance. if I comment out most of the code in SetupDefaultAddressSpace(), the memory stays low. I will look into it a little deeper

nauful, 2022-02-03:
If you want to work with an embedded device and using the standard UA address space isn't important (i.e. type nodes, server nodes, buildinfo nodes, etc), you can just skip that method entirely and require very little memory for your address table.

closed 2022-02-03 by nauful

pspeybro, 2022-02-07:
Hi @nauful, just wanted to let you know that with some refactoring, this memory spike can be significantly reduced.
I just moved node creation for the default namespace to a method that takes the necessary parameters which already reduced the spike to less than a 3rd. Splitting up the address space generation reduced it even further.

This results in the memory only going up to around 25MB during startup. Lower than the memory when a connection is made.
Because I was just looking for a quick and dirty solution, the code change is quite rudimentary and I'm sure there are much more efficient ways to tackle this, but this resolves my issue: https://github.com/pspeybro/LibUA/commit/4e7fa5478e010de48a260057b8a5fa0eff865ce2