nauful/LibUA #74: How does the server change the label value

nihao-git, 2022-02-11:
Hello, How does the server change the label value，I've assigned permissions to each label, but I still can't write to the client.

nauful, 2022-02-11:
I don't understand what label you are referring to. Could you show me a screenshot or code sample?

nihao-git, 2022-02-11:
[image]
Sorry, I checked yesterday, it should be called node in your code, the value of node cannot be written from the client side, I have added permission to the node in the server code, but it still cannot be written, is it my wrong to add permission, or should not be added in this way, please advise, thank you.

nauful, 2022-02-11:
Server nodes don't accept UA writes by default. To do this, in your application (subclass of `LibUA.Server.Application`), you can implement this method:
`public virtual UInt32[] HandleWriteRequest(object session, WriteValue[] writeValues)`

You can match node ids according to `WriteValue.NodeId` and Value would be `WriteValue.AttributeId`. If you update the value of a node, remember to call `MonitorNotifyDataChange` to send a notification to all subscribers.

nihao-git, 2022-02-14:
Thank you, problem solved.

closed 2022-02-14 by nihao-git