nauful/LibUA #76: Can not Write to value on server

Gfreeddos, 2022-03-01:
Hello. 
I am start you Server opc and connect from OPC Windows Client (not your client). 
Try change value node and take error BadWriteNotSupported = 0x80730000 from client. 
How change create value node on server? Or handler?
Sorry i am just learn C# programing.

nauful, 2022-03-02:
In your appliction class, you need to implement the following method:
```			public virtual UInt32[] HandleWriteRequest(object session, WriteValue[] writeValues)
			{
				var respStatus = new UInt32[writeValues.Length];
				for (int i = 0; i < writeValues.Length; i++)
				{
					respStatus[i] = (UInt32)StatusCode.BadNotWritable;
				}

				return respStatus;
			}```
Otherwise the default is BadNotWritable.

closed 2022-03-02 by nauful

Gfreeddos, 2022-03-03:
Ok. I Use this method (override HandleWriteRequest).
Take callback when change Value, but value not save. 
How can I use this HandleWriteRequest for write value?

nauful, 2022-03-04:
You would have to store the written value somewhere, then return it in `HandleRead`:
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestServer/Program.cs#L208

When storing the written value, you should also publish a data change notification (publishes new value) using `MonitorNotifyDataChange`:
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestServer/Program.cs#L422