nauful/LibUA #80: I would need help with monitoreditems please

ne-fas-tus, 2022-04-27:
Not knowing where to contact you I do it here. I have already written to you in a previous report that I would like to communicate with an OPC-UA server of a Modicon m241 PLC from a client application made in Unity3D. I imported your library and managed to connect to the server. I can activate a session, a subscription, browse the variables, read the values, write them etc... as written in your sample program. What I would like to do now is to use the MonitoredItems but I did not understand how it works. After I call the CreateMonitoredItems method to monitor a variable, what should I do? Should I receive NotificationMessages from the server from which to extract the variable? How do I read them? I don't understand how I have to do it. Sorry if I'm so ignorant! If you could write a few lines of example, I would be infinitely grateful

nauful, 2022-04-27:
As I understand it, you have a client and you want to subscribe to server nodes.

Start by creating a subscription:
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestClient/Program.cs#L239

Create a client handle for the nodes you want to monitor (user-defined, I just use 1, 2, 3... in the sample):
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestClient/Program.cs#L249

Create some monitored items for this subscription:
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestClient/Program.cs#L252
Specifically, you probably just want to monitor `NodeAttribute.Value`:
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestClient/Program.cs#L260
Note how your user-defined client handles are passed in (e.g. first monitored item uses first user-defined handle=1, second has handle=2, etc).
`monitorCreateResults` should give you some helpful information.

Override `NotifyDataChangeNotifications`:
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/TestClient/Program.cs#L79
This method is now called whenever a change for those monitored items comes back. You receive `subscriptionId` and a sequence of `user-defined handles -> notifications`.

Try stepping through the TestServer and TestClient projects if you get stuck. TestServer publishes data once per second, TestClient subscribes to 3 of those nodes and prints updates to the console.

ne-fas-tus, 2022-04-27:
Solved! Then, it also works with MonitoringMode.Reporting. The problem was that I was getting the ID of the node to be monitored wrong but I got GOOD from the CreateMonitoredItems method. I logged the statusCode of the monitorsCreateResults, it was BadIdParameter... Now the NotifyDataChangeNotifications method is called and returns the updated value correctly. However, I had not understood the operation well before I explained it to myself. Thank you thank you thank you

nauful, 2022-04-27:
Glad to hear it worked.

closed 2022-04-27 by nauful