nauful/LibUA #84: Question: Array of complex types

generik0, 2022-07-02:
HI. Thanks for your awesome lib!

I have a model (S88):

```c#
public class Command
{
    public string UnitMode { get; set; }
    public bool UnitModeChangeRequest { get; set; }
    
    public string CntrlCmd { get; set; }
    public string CmdChangeRequest { get; set; }
}
```

And I need an array of 10 of them. Can you provide a hint as to the right direction, please?
Or would you recommend not using arrays, and having individual objects?

nauful, 2022-07-09:
You might use the `MemoryBuffer` class to serialize your instances into a byte array, then set that as the body for an `ExtensionObject`. `DataValue` can accept that `ExtensionObject` instance with your serialized data as the value.

generik0, 2022-07-22:
thx. i'll try

closed 2022-07-22 by generik0