nauful/LibUA #85: OPC node value update event

fch96733357, 2022-07-11:
Hi, 
      I am using LibUA .net library to integrate with my database system, I am able to create nodes from database, and update nodes value from my application using socket; I just wonder how to create write request event for my nodes (read&write), when client is updating (set new value) the node's value, I am able to capture the new value so that I can pass the new value to my application. Something like OnValuechanged(node mnode,int newvalue).

Thanks and best regards
Feng Chonghai
Fch96733357@gmail.com

nauful, 2022-07-11:
Try implementing `public virtual UInt32[] HandleWriteRequest(object session, WriteValue[] writeValues)` in your server `Application`.  This will notify you of any values written over UA to your database.

fch96733357, 2022-07-12:
Nauful, Thank you very much for quick response.

      Yes, I did try to implement HandleWriteRequest virtual function, however, when I tried to write new value via OPC UA client, debug windows shown below
The new value does not in writeValues[], is it this correct?

Best Regards
Feng

From: nauful
Sent: 2022年7月12日 4:32
To: nauful/LibUA
Cc: fch96733357; Author
Subject: Re: [nauful/LibUA] OPC node value update event (Issue #85)

Try implementing public virtual UInt32[] HandleWriteRequest(object session, WriteValue[] writeValues) in your server Application. This will notify you of any values written over UA to your database.

nauful, 2022-07-12:
Hello Feng,

Could you please paste the screenshot in Github? I don't see any image if you replied via email.

fch96733357, 2022-07-14:
Hi Nauful,

     Thanks for your great OPC lib, it solved my problem. Thanks a lot

fch96733357, 2022-07-14:
Close

closed 2022-07-15 by nauful

fch96733357, 2022-10-11:
Thanks Nauful, I have managed to get it work, because I did not initialize value.

Best Regards
Feng

From: Feng ChongHai
Sent: 2022年7月12日 8:19
To: nauful/LibUA; nauful/LibUA
Cc: Author
Subject: RE: [nauful/LibUA] OPC node value update event (Issue #85)

Nauful, Thank you very much for quick response.

      Yes, I did try to implement HandleWriteRequest virtual function, however, when I tried to write new value via OPC UA client, debug windows shown below
The new value does not in writeValues[], is it this correct?

Best Regards
Feng

From: nauful
Sent: 2022年7月12日 4:32
To: nauful/LibUA
Cc: fch96733357; Author
Subject: Re: [nauful/LibUA] OPC node value update event (Issue #85)

Try implementing public virtual UInt32[] HandleWriteRequest(object session, WriteValue[] writeValues) in your server Application. This will notify you of any values written over UA to your database.