nauful/LibUA #86: Publication Interval is halved

ne-fas-tus, 2022-07-11:
Hi nauful.
Given that your library is great and works great, I encountered a problem regarding the publishing interval. I think it's a server-side problem. I create client and server with your library. When I create a subscription in the client, I pass requestedPublishInterval to the CreateSubscription method but the server sends me the notifications twice as often, as if I had requested half the publication interval. Can you check if it does to you too? I didn't change the methods you created, so it's strange.
Thank you

ne-fas-tus, 2022-07-13:
My notification type is set to `AtPublish`. 
I try debbuging GetNextPendingSubscription. `RequestedPublishingInterval` is correct, but `PublishInterval` is halved.
I noticed that in this line 
https://github.com/nauful/LibUA/blob/7c9e1a22ba38743af4dbdd3bdb1420788e2238e6/NET%20Core/LibUA/NetDispatcher.cs#L2909

the publish interval is obtained from the division of the revisedPublishInterval. I tried to edit line with
```C#
PublishInterval = TimeSpan.FromMilliseconds(revisedPublishInterval);
```
 and fix. But I don't know if I changed something I shouldn't have.

nauful, 2022-07-13:
Oops, that's a bug. That should only be for keep-alive intervals.

Pushed a change, please try now.

ne-fas-tus, 2022-07-13:
It works!!

closed 2022-07-15 by nauful