nauful/LibUA #88: Updated nuget package?

smitha-cgi, 2022-07-19:
Latest package available via nuget is 1.0.1 dated July 11 2020. Any chance you could publish a newer version? Would rather refer to a package than have to clone the repo and store the source with the rest of my app code.

nauful, 2022-07-21:
You might be looking at a fork. The main branch has never published a nuget package. I will consider this when I have time to look at it.

closed 2022-07-21 by nauful

smitha-cgi, 2022-07-21:
Ah, no worries. The current package is pointing to your github page as the project website, so I guess davidguidali published it without your knowledge a few years ago.

[image]