nauful/LibUA #92: Bug decoding null Arrays

wckivX, 2022-08-19:
When receiving a CreateMonitoredItems response from a FactoryTalk Linx Gateway OPC UA server, it sent a null array.  The client attempted to create and decode an array of 4,294,967,295 elements instead of a null Array.  The OPC UA Online Reference (2022) OPC 10000-6 section 5.2.5 states that array lengths are to be encoded as Int32 values.  The code in Client.cs and MemoryBufferExtensions.cs is encoding array lengths as UInt32.  On the wire both Int32 and UInt32 are the same 4 bytes.  When decoding the code sometimes checks for 0xFFFFFFFFu (-1) and handles a null array correctly, but does a poor job of it in Client.cs.

I think it would be clearer it encoded Array lengths as Int32.  C# Array.Length property is a signed integer.

Also, when throwing generic System.Exceptions for TODO branches (MemoryBufferExtensions),  it would be nice to include a message text that could be searched for in the source code.

I can provide patched code.

nauful, 2022-08-21:
Makes sense. If you create a PR I can review and merge in.

nauful, 2022-08-23:
Fixed https://github.com/nauful/LibUA/commit/2d67a2686ff14aa9092c91ab28f762d05c383fc4

closed 2022-08-23 by nauful