nauful/LibUA #97: BadIdentityTokenInvalid

DerksenA, 2022-08-30:
Hi nauful.

First of all, great work. 
I already integratet your lib with a PLC and it's working. 
Now I have a Codesys Soft SPS and would like to connect to. When using Anonymous everything is working. But I want to connect with Username and Password.
Here's my code:
`var appDesc = new ApplicationDescription(
				"urn:DemoApplication", "uri:DemoApplication", new LocalizedText("UA SDK client"),
				ApplicationType.Client, null, null, null);

			ApplicationDescription[] appDescs = null;
			EndpointDescription[] endpointDescs = null;

			//var client = new DemoClient("192.168.1.7", 7718, 10);
			var client = new DemoClient("192.168.179.200", 4840, 1000);
			client.Connect();
			client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);
			client.FindServers(out appDescs, new[] { "en" });
			client.GetEndpoints(out endpointDescs, new[] { "en" });
			//client.Disconnect();

			// Check matching message security mode and security policy too
			// Lazy way to find server certificate is just grab any endpoint with one
			byte[] serverCert = endpointDescs
				.First(e => e.ServerCertificate != null && e.ServerCertificate.Length > 0)
				.ServerCertificate;

			var usernamePolicyDesc = endpointDescs
				.First(e => e.UserIdentityTokens.Any(t => t.TokenType == UserTokenType.UserName))
				.UserIdentityTokens.First(t => t.TokenType == UserTokenType.UserName)
				.PolicyId;

			// Create new client object to reset previous secure channel settings
            client = new DemoClient("192.168.179.200", 4840, 1000);
			var connectRes = client.Connect();
            
			client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, serverCert);
			//var openRes = client.OpenSecureChannel(MessageSecurityMode.None, SecurityPolicy.None, null);
			var createRes = client.CreateSession(appDesc, "urn:DemoApplication", 120);
            //var activateRes = client.ActivateSession(new UserIdentityAnonymousToken("Anonymous"), new[] { "en" });

            new UserIdentityTokenType();
            //client.ActivateSession(new UserIdentityUsernameToken(usernamePolicyDesc,))
            
            var activateRes = client.ActivateSession(
                new UserIdentityUsernameToken(usernamePolicyDesc, "abc",
                    (new UTF8Encoding()).GetBytes("abc"), Types.SignatureAlgorithmRsa15),
                new[] { "en" });`

Using another Client is working.
On Point activateRes I get the Error BadUnknownResponse.
I also used Wireshark to figure out the problem but have no idea what to do now.
Here's a snipped from wireshark and also the files attached:  

[image]
[Wireshark opc.zip](https://github.com/nauful/LibUA/files/9455045/Wireshark.opc.zip)
 
Thanks in advance.

nauful, 2022-09-01:
Hello,

The other client used RSA-OAEP for encrypting the password, but the code sample uses RSA-1.5 from `Types.SignatureAlgorithmRsa15`, which I would guess your server doesn't accept. I don't have any plans to add more encryption methods right now, and I don't currently support RSA-OAEP. You might try an unencrypted password but switch your connection to signed & encrypted.

closed 2022-09-01 by nauful

RubenBuitendijk, 2023-01-13:
Hello Derksen, I have exactly the same issue and i need full encryption for my OPC. Did you find a solution?

Halowek, 2023-01-13:
Hello Derksen, Did you find a solution yet? I have exactly the same issue and i can't get it working. I need the full encryption. On anonymous it works fine.

nauful, 2023-01-16:
Implemented more security policies and user identity token signatures (rsa-oaep) here:
https://github.com/nauful/LibUA/commit/6534b413b5abaca258542aabd81cf85094c18838