nauful/LibUA #104: Browsing from client

dstickel, 2022-10-18:
I am using the BrowseDescription method in the client class. I have multiple treenode tiers in the server but it only returns the root object and the next tier of objects. I noticed the ContinuationPoint is returning null which should be used for the 

while (browseVariables[0].ContinuationPoint != null) 

to browse lower tier objects. Is the ContinuationPoint code implemented? I tested this with some third party clients and they return all tiers of objects.

nauful, 2022-10-18:
I'm not sure if that has been implemented completely (probably not). If you can share a test server with me, I can implement the remainder.

Btw, if you're using this library in a project, I would like to ask permission to add your use cases and organization to my list of users.

dstickel, 2022-10-18:
Here is a copy of the project. It has an Allen Bradley driver I wrote in it that I am working to implement. This is POC to see if I can get everything working like I hope to. I created some custom methods for adding the drivers, tags, and folders into the address space. I very well could have something wrong here. I attached a screenshot of a connection using Unified Automations client.

I am not using this in a project yet but hope to. If I get everything working as I want then you have my permission for anything you would like. The ultimate goal is to use it for a new Scada software I am creating. I hate how bloated and large the OPC UA standard is so your version is exactly what I would like to have.

Really appreciate any help you could give me with this.

dstickel, 2022-10-18:
[image]

dstickel, 2022-10-18:
Here is a screenshot of the Unified Automation Client connecting to it. I emailed a copy of the project.

nauful, 2022-10-18:
I don't think github allows emailing attachments. You can email me directly: username @gmail.com

dstickel, 2022-10-18:
[LibUA.zip](https://github.com/nauful/LibUA/files/9815349/LibUA.zip)

dstickel, 2022-10-18:
Above is the project.

nauful, 2022-10-18:
You shouldn't browse into lower tiers by `ContinuationPoint`. When you browse the top-level node, look at the references:
```
-		browseResults[0]	{LibUA.Core.BrowseResult}	LibUA.Core.BrowseResult
		ContinuationPoint	null	byte[]
-		Refs	{LibUA.Core.ReferenceDescription[6]}	LibUA.Core.ReferenceDescription[]
-		[0]	{LibUA.Core.ReferenceDescription}	LibUA.Core.ReferenceDescription
+		BrowseName	{[0] Objects}	LibUA.Core.QualifiedName
+		DisplayName	{[] Objects}	LibUA.Core.LocalizedText
		IsForward	false	bool
		NodeClass	Object	LibUA.Core.NodeClass
+		ReferenceTypeId	{ns=35;s=MachineView}	LibUA.Core.NodeId
+		TargetId	{ns=0;i=85}	LibUA.Core.NodeId
+		TypeDefinition	{ns=0;i=61}	LibUA.Core.NodeId
-		[1]	{LibUA.Core.ReferenceDescription}	LibUA.Core.ReferenceDescription
+		BrowseName	{[0] Alarms}	LibUA.Core.QualifiedName
+		DisplayName	{[] Alarms}	LibUA.Core.LocalizedText
		IsForward	true	bool
		NodeClass	Object	LibUA.Core.NodeClass
+		ReferenceTypeId	{ns=0;i=35}	LibUA.Core.NodeId
+		TargetId	{ns=2;i=1}	LibUA.Core.NodeId
+		TypeDefinition	{ns=0;i=0}	LibUA.Core.NodeId
-		[2]	{LibUA.Core.ReferenceDescription}	LibUA.Core.ReferenceDescription
+		BrowseName	{[0] Data}	LibUA.Core.QualifiedName
+		DisplayName	{[] Data}	LibUA.Core.LocalizedText
		IsForward	true	bool
		NodeClass	Object	LibUA.Core.NodeClass
+		ReferenceTypeId	{ns=0;i=35}	LibUA.Core.NodeId
+		TargetId	{ns=2;i=2}	LibUA.Core.NodeId
+		TypeDefinition	{ns=0;i=0}	LibUA.Core.NodeId
-		[3]	{LibUA.Core.ReferenceDescription}	LibUA.Core.ReferenceDescription
+		BrowseName	{[0] Historical}	LibUA.Core.QualifiedName
+		DisplayName	{[] Historical}	LibUA.Core.LocalizedText
		IsForward	true	bool
		NodeClass	Object	LibUA.Core.NodeClass
+		ReferenceTypeId	{ns=0;i=35}	LibUA.Core.NodeId
+		TargetId	{ns=2;i=3}	LibUA.Core.NodeId
+		TypeDefinition	{ns=0;i=0}	LibUA.Core.NodeId
-		[4]	{LibUA.Core.ReferenceDescription}	LibUA.Core.ReferenceDescription
+		BrowseName	{[0] Methods}	LibUA.Core.QualifiedName
+		DisplayName	{[] Methods}	LibUA.Core.LocalizedText
		IsForward	true	bool
		NodeClass	Object	LibUA.Core.NodeClass
+		ReferenceTypeId	{ns=0;i=35}	LibUA.Core.NodeId
+		TargetId	{ns=2;i=4}	LibUA.Core.NodeId
+		TypeDefinition	{ns=0;i=0}	LibUA.Core.NodeId
-		[5]	{LibUA.Core.ReferenceDescription}	LibUA.Core.ReferenceDescription
+		BrowseName	{[0] Pubsub}	LibUA.Core.QualifiedName
+		DisplayName	{[] Pubsub}	LibUA.Core.LocalizedText
		IsForward	true	bool
		NodeClass	Object	LibUA.Core.NodeClass
+		ReferenceTypeId	{ns=0;i=35}	LibUA.Core.NodeId
+		TargetId	{ns=2;i=5}	LibUA.Core.NodeId
+		TypeDefinition	{ns=0;i=0}	LibUA.Core.NodeId
		StatusCode	0	uint
```

`ReferenceTypeId` with ns=0;i=35 means `RefType.Organizes`. UaExpert also shows the children are browsed via references:
[image]

To browse the hierarchy, call Browse on the root then recursively call Browse on `TargetIds` given by references of the reference type `RefType.Organizes`.

`ContinuationPoint` is when the Browse results for that particular node are too large to fit in a single message response.

dstickel, 2022-10-18:
Do you have an example of that? I'm trying a few things now.

nauful, 2022-10-18:
This should enumerate everything under the server's objects node:
```
Queue<NodeId> nodeQueue = new Queue<NodeId>();
nodeQueue.Enqueue(new NodeId(2, (uint)UAConst.ObjectsFolder));
while (nodeQueue.TryDequeue(out NodeId currentNode))
{
... store currentNode somewhere

	client.Browse(new BrowseDescription[]
	{
		new BrowseDescription(
			currentNode,
			BrowseDirection.Forward,
			NodeId.Zero,
			true, 0xFFFFFFFFu, BrowseResultMask.All)
	}, 10000, out BrowseResult[] childrenBrowseResults);

	foreach (var reference in childrenBrowseResults[0].Refs)
	{
		if (reference.ReferenceTypeId.EqualsNumeric(0, (uint)RefType.Organizes))
		{
			nodeQueue.Enqueue(reference.TargetId);
		}
	}
}
```

dstickel, 2022-10-19:
Debugging it gets to

foreach (var reference in childrenBrowseResults[0].Refs)
                {
                    if (reference.ReferenceTypeId.EqualsNumeric(0, (uint)RefType.Organizes))
                    {
                        nodeQueue.Enqueue(reference.TargetId);
                    }
                }
then jumps out as there is no childrenBrowseResults[0].Refs. It never iterates through anything.

nauful, 2022-10-19:
My mistake, the namespace index should be 0:
`nodeQueue.Enqueue(new NodeId(0, (uint)UAConst.ObjectsFolder));`

nauful, 2022-10-19:
I would guess you have circular references. Please check the children references have `OrganizedBy` and not `Organized` for the reference of child->parent. If that doesn't work I can look at the server code again.

dstickel, 2022-10-19:
I am trying some different code and seeing that now. They are set to organize. I will change that and try it again. Thanks.

nauful, 2022-10-19:
Hmm, that should have worked with inverse. Try checking if `IsForward` is set on the returned reference and skip the ones with false.

dstickel, 2022-10-19:
Hhhmm.... I tried it like this and it appears to be working now. I haven't tried it in the TreeNode yet just in a Console.WriteLine.

BrowseResult[] childrenBrowseResults = null;
			Queue<NodeId> nodeQueue = new Queue<NodeId>();
			nodeQueue.Enqueue(new NodeId(0, (uint)UAConst.ObjectsFolder));
			while (nodeQueue.TryDequeue(out NodeId currentNode))
			{
				// store currentNode somewhere
			
	         client.Browse(new BrowseDescription[]
	          {
		       new BrowseDescription(
			   currentNode,
			   BrowseDirection.Forward,
			   NodeId.Zero,
			   true, 0xFFFFFFFFu, BrowseResultMask.All)
	          }, 10000, out childrenBrowseResults);

				foreach (var reference in childrenBrowseResults[0].Refs)
				{
					if (reference.ReferenceTypeId.EqualsNumeric(0, (uint)RefType.Organizes))
					{
						nodeQueue.Enqueue(reference.TargetId);
					}
					Console.WriteLine(reference.DisplayName);
				}
			}

            for (int i = 0; i < childrenBrowseResults[0].Refs.Length; i++)
            {
                Console.WriteLine(childrenBrowseResults[0].Refs[i].DisplayName);
            }

dstickel, 2022-10-19:
That did work. I just have to figure out how to parse it with the TreeNode correctly but it is returning the data I needed. Thank you for the help. Very appreciated!

nauful, 2022-10-19:
You're welcome. Let me know if you could use any more help.

closed 2022-10-19 by nauful