nauful/LibUA #106: Alarm Interactions Via Client

dstickel, 2022-10-27:
Has it been implemented to interact with alarms from the client? Is NotifyEventNotifications used for this? Examples would be acknowledging the alarm.

Is creating a client.HistoryRead using ReadEventDetails and HistoryReadValueId, like in the sample client, required for using alarms or what would be required if not? I know you need the EventNotification but is there a generic ReadValueId that can be used for the MonitoredItemCreateRequest that will observe all alarms on the client side?

nauful, 2022-10-27:
There are alarms as part of the test server, but I haven't done anything for alarms on the client side. The last time I tried, I couldn't get anything generic to work similarly across different server vendors.

dstickel, 2022-10-27:
Ok thanks.

closed 2022-10-28 by nauful