nauful/LibUA #109: Closed Connection on bad quality status

game4automation, 2022-12-23:
Hi, 
in a test setup where one subscribed node is giving back a bad quality status the connection is automatically closed by the Method ThreadTarget in Client.cs even if other nodes are still working as expected. But I would like to keep up the connection up and running. Do you know how to do that? Is it a bug in Client.cs?

Best regards 
Thomas

nauful, 2022-12-23:
Hello Thomas,

Reading a node with bad quality status should not close the connection. Connections are automatically closed when reading an invalid message (decoding error) but I may be wrong. Otherwise the client only disconnects when you call the method. Please upload a Wireshark capture and I can investigate further.

By the way, if you are using this library for any projects, there is no obligation but I would be very interested in hearing where and how. I would appreciate your permission to use that information in my list of use cases. If you prefer email, my email is my username @gmail.com

Thanks.

game4automation, 2022-12-27:
Hi,
we are using your library in our Digital Twin Solution for Unity (see https://realvirtual.io). You can reference to our solution if you would like.

I am not a Wireshark expert so I am describing you here how to reproduce the problem.

Our test opcua server is the C++ demo server from Unified automation which you could get here:
https://www.unified-automation.com/de/downloads/opc-ua-servers.html

In the demo server we subscribe to the node 
ns=2;s=Demo.Dynamic.Scalar.Quality_DynamicValue

Only when subscribing to this node the conneection is closed by ThreadTarget in Client.cs. 

Thanks a lot
Thomas

nauful, 2022-12-27:
Great, thanks!

I don't have login for unified-automation, would you be able to upload that in a zip file here?

game4automation, 2022-12-27:
I have send you a download link to your mail. Thanks a lot.

nauful, 2022-12-27:
Hello Thomas,

Looks like an exception in the demo app in the callback `NotifyDataChangeNotifications`. Please change to:
```
public override void NotifyDataChangeNotifications(uint subscrId, uint[] clientHandles, DataValue[] notifications)
{
	for (int i = 0; i < clientHandles.Length; i++)
	{
		Console.WriteLine("subscrId {0} handle {1}: {2}", subscrId, clientHandles[i], notifications[i].Value?.ToString());
	}
}
```

The issue was that node sometimes publishes `notifications[i].Value` as null. Now I see this output when subscribed to only that node:
```
subscrId 1368329684 handle 1: 6532.900000000567
subscrId 1368329684 handle 1: 6534.0000000005675
subscrId 1368329684 handle 1: 6535.100000000568
subscrId 1368329684 handle 1: 6536.200000000568
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1: 6543.900000000571
subscrId 1368329684 handle 1: 6545.000000000571
subscrId 1368329684 handle 1: 6546.1000000005715
subscrId 1368329684 handle 1: 6547.200000000572
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1:
subscrId 1368329684 handle 1: 6554.900000000574
subscrId 1368329684 handle 1: 6556.000000000575
subscrId 1368329684 handle 1: 6557.100000000575
subscrId 1368329684 handle 1: 6558.2000000005755
...
```

While not robust in the sample application, you would need to handle exceptions during callbacks so they don't bubble up to the client's connection thread. Let me know if that works.

Thanks,
Nauful

game4automation, 2022-12-28:
Hi, thanks a lot - your hint solved the issue.

nauful, 2022-12-28:
Great, let me know if anything else comes up.

closed 2022-12-28 by nauful