nauful/LibUA #111: BadUnknownResponse on CreateSession with Username&Password and encrypted connection

fraluderin, 2023-01-04:
First of all, thank you for this amazing library. It is my final choose after searching the internet for months looking for a good and easy to use OPC UA library.

I'm experiencing an issue using the library in order to connect to a Siemens PLC.
Everything works fine until I change connection mode.
With anonymous token and no security policy / message encryption, I have no issues. If I enable _SecurityPolicy_ (Basic256Sha256) and _MessageSecurityMode_ (SignAndEncrypt), I get **BadUnknownResponse** status code on CreateSession().
Connecting with UA Expert gives me no issues (same connection parameters, same server).

Here's my code (it is nearly unchanged from provided TestClient):

```
  private void InternalConnect2()
        {
            bool usernameAndPassword = true;
            bool secureConnection = true;

            var appDesc = new ApplicationDescription("urn:DemoApplication", "uri:DemoApplication", new LocalizedText("UA SDK client"), ApplicationType.Client, null, null, null);

            var connectres1 = InternalClient.Connect();
            InternalClient.OpenSecureChannel(LibUA.Core.MessageSecurityMode.None, LibUA.Core.SecurityPolicy.None, null);
            InternalClient.FindServers(out ApplicationDescription[]? appDescs, new[] { "en" });
            InternalClient.GetEndpoints(out EndpointDescription[]? endpointDescs, new[] { "en" });
            InternalClient.Disconnect();

            byte[] serverCert = endpointDescs.First(e => e.ServerCertificate != null && e.ServerCertificate.Length > 0).ServerCertificate;
            string usernamePolicyDesc;
            if (usernameAndPassword)
                usernamePolicyDesc = endpointDescs.First(e => e.UserIdentityTokens.Any(t => t.TokenType == UserTokenType.UserName)).UserIdentityTokens.First(t => t.TokenType == UserTokenType.UserName).PolicyId;
            else
                usernamePolicyDesc = endpointDescs.First(e => e.UserIdentityTokens.Any(t => t.TokenType == UserTokenType.Anonymous)).UserIdentityTokens.First(t => t.TokenType == UserTokenType.Anonymous).PolicyId;

            var connectRes = InternalClient.Connect();

            StatusCode openRes;
            if (secureConnection)
                openRes = InternalClient.OpenSecureChannel(LibUA.Core.MessageSecurityMode.SignAndEncrypt, LibUA.Core.SecurityPolicy.Basic256Sha256, serverCert); // No issues
            else
                openRes = InternalClient.OpenSecureChannel(LibUA.Core.MessageSecurityMode.None, LibUA.Core.SecurityPolicy.None, null); // No issues

            var createRes = InternalClient.CreateSession(appDesc, "urn:DemoApplication", 1000); // Returns BadUnknownResponse if secureConnection is used; no issues if secureConnection is disabled

            StatusCode activateRes;
            if (usernameAndPassword)
                activateRes = InternalClient.ActivateSession(new UserIdentityUsernameToken(usernamePolicyDesc, "user", new UTF8Encoding().GetBytes("psw"), Types.SignatureAlgorithmSha256), new[] { "it" }); // Returns BadUnknownResponse if secureConnection is used
            else
                activateRes = InternalClient.ActivateSession(new UserIdentityAnonymousToken(usernamePolicyDesc), new[] { "en" }); // Returns BadUnknownResponse if secureConnection is used

        }
```

In addition, if I disable secureConnection (the bool flag has testing purpose), the CreateSession() cause no errors, but I get _BadUnknownResponse_ on ActivateSession().

The library is making an amazing job with my experimental project, it would be a nightmare rewriting it all using another solution - assuming there's something else like LibUA on the internet ;)

Probably there's something I'm doing wrong, so thank you in advance for your help.

Attached UA Expert log file obtained connecting to server and a capture from Siemens Tia Portal PLC configuration.

[image]
[ua expert.log](https://github.com/nauful/LibUA/files/10342987/ua.expert.log)

nauful, 2023-01-04:
Hello,

Always happy to hear when others enjoy this library. Just to confirm your setup: you're connecting from a LibUA client to your PLC server?

What happen if, without username and password, you try:
- Basic128Rsa15 Sign & Encrypt
- Basic256 Sign & Encrypt
- Basic256Sha256 Sign&Encrypt

I suspect trouble with the username and password while using connection encryption:
https://github.com/nauful/LibUA/issues/97#issuecomment-1234871858
https://github.com/nauful/LibUA/issues/34#issuecomment-1293996630

fraluderin, 2023-01-04:
Hi, I tried what you suggested.

All three options returned BadUnknownResponse (both from CreateSession() and ActivateSession()) if used in combination with AnonymousToken (no username and password).

If I switch to no encryption (SecurityPolicy = None and MessageSecurityMode = None) using username and password, CreateSession() returns Good but ActivateSession() still returns BadUnkownResponse.

nauful, 2023-01-05:
Let me investigate, will get back to you.

Can you upload a Wireshark capture of no security using username and password? I want to see the traffic where `ActivateSession` fails.

nauful, 2023-01-05:
After some testing, I would guess the OPC UA server is rejecting the client's certificate.

fraluderin, 2023-01-07:
Hi nauful, attached the file you requested.
I tried to change device just in case this was an issue with that specific PLC.
This time I used another Siemens PLC (IP address 192.168.7.118). I configured the PLC:
- to accept any client certificate during runtime;
- installed the client certificate created by LibUA.
Have you ever worked with Siemens PLC? Do you know if there's something I may did wrong?

Thank you for your support!

[password only.zip](https://github.com/nauful/LibUA/files/10366050/password.only.zip)

nauful, 2023-01-08:
It looks like the server returns this error:
`ActivateSessionRequest -> ServiceFault: 0x80200000 [BadIdentityTokenInvalid]`
However I don't get this error from the OPC Foundation's demo C++ server, so I think this is a configuration issue. If it works with another client, probably some field is sent, not sent, or has a different value. Can you show me the same capture but with UaExpert or another client which works? The one above has encrypted so I cannot see `ActivateSessionRequest`.

I have used Siemens PLC in the past and it was generally fine, but your versions may be different. In any case, I only worked with traffic analysis to mimic requests and responses for common vendors, so compatibility comes from analyzing network dumps over specific implementations. Seeing `ActivateSessionRequest` unencrypted with a user login from a working client should reveal the issue.

By the way, if you are using this library for any projects, there is no obligation but I would be very interested in hearing where and how. I would appreciate your permission to use that information in my list of use cases. If you prefer email, my email is my username @gmail.com

nauful, 2023-01-16:
Implemented more security policies and user identity token signatures (rsa-oaep) here:
https://github.com/nauful/LibUA/commit/6534b413b5abaca258542aabd81cf85094c18838

Let me know if you have any problems.

closed 2023-01-16 by nauful