nauful/LibUA #114: Adress like opc.tcp://172.XXX.XXX.XXX:4880/Staubli

game4automation, 2023-02-07:
Hi, I am wondering how we could access an opcua server with the client on an adress like opc.tcp://172.XXX.XXX.XXX:4880/Staubli. I don't see a parameter besides the IP Adress to configure this adress on the connection. Do you have any hint?

Thanks a lot
Thomas

nauful, 2023-02-07:
Could you connect to that via UaExpert with security mode set to None and share a Wireshark capture?

I wonder if it's just a node off the root to change your view of the address space and not actually part of the connection process.

eliovalenzuela, 2023-02-08:
Hello, here is the Wireshark capture. The server address is 192.168.31.99, and the UaExpert client is at 192.168.31.184.
[UAExpert_CS9.zip](https://github.com/nauful/LibUA/files/10687790/UAExpert_CS9.zip)

nauful, 2023-02-08:
For DemoClient, you can use the second constructor option which includes path ("Staubli"):
```
public DemoClient(string Target, int Port, string Path, int Timeout)
	: base(Target, Port, Path, Timeout)
{
	LoadCertificateAndPrivateKey();
}
```

This should handle CreateSessionRequest and pass that case. Let me know if that works.

closed 2023-02-16 by game4automation