nauful/LibUA #115: Connecting to Mitsubishi DxpServer with Username and Password

game4automation, 2023-02-16:
Hi, I am trying to connect to the DxpServer with Username and Password.

Anonymous works. But when connecting with Username and password I am failing. I tried different SignatureAlgorithm.

  var usernamePolicyDesc = endpointDescs
                        .First(e => e.UserIdentityTokens.Any(t => t.TokenType == UserTokenType.UserName))
                        .UserIdentityTokens.First(t => t.TokenType == UserTokenType.UserName)
                        .PolicyId;
                    activateRes = client.ActivateSession(
                        new UserIdentityUsernameToken(usernamePolicyDesc, UserName,
                            new UTF8Encoding().GetBytes(Password), LibUA.Core.Types.SignatureAlgorithmRsa15),
                        new[] {"en"});

There is no SecurityPolicy / SecurityMode set on the Server. Just a password and username should be needed.

Any ideas?

Thanks
Thomas

nauful, 2023-02-16:
Hello Thomas,

In security policy and mode as None, null is sent for user name signature algorithm (specified value is ignored).

If you are able to make this work with a normal UA client, share a Wireshark dump and I can look at the difference.

game4automation, 2023-02-17:
Hi, 

thanks a lot.

I was able to get it up running with 
             activateRes = client.ActivateSession(
                        new UserIdentityUsernameToken(usernamePolicyDesc, UserName,
                            new UTF8Encoding().GetBytes(Password),LibUA.Core.Types.SignatureAlgorithmRsaOaep),

How can I detect which type of user name signature algorithm I should use for connection? Is it given back somehow from the server?

nauful, 2023-02-17:
Glad to hear it worked. Look at the security policy uri for the appropriate `UserIdentityToken` (`UserTokenType` == `UserName`) in the `EndpointsDescription`:
[image]

`PolicyId` is custom (I use numbers), but `SecurityPolicyUri` is a constant (look in the library source for that exact value).

closed 2023-02-17 by nauful

game4automation, 2023-02-17:
Hi it is still strange to me.

Server gives me back http://opcfoundation.org/UA/SecurityPolicy#Basic256 but login needs to be made with Types.SignatureAlgorithmRsaOaep which is "http://www.w3.org/2001/04/xmlenc#rsa-oaep"

Do you know why. Is there any change to select automatically what is needed from server side?

Thanks a lot for your work - it is really great.

Thomas

nauful, 2023-02-18:
Glad to hear you find this library useful.

Can you share a Wireshark capture of the response to browsing endpoints? It may be an error on the server's part. Otherwise, if you look at the endpoint description with the same `UserTokenType` and `MessageSecurityMode`, you should get the correct `SecurityPolicyUri` if one exists (no match if it does not exist).

As a compability mechanism you can simply try to enumerate all of the options while connecting and take the first which isn't rejected.