nauful/LibUA #116: Monitored item delivers errors in clients log

stefanofraccaro, 2023-03-02:
When you monitor an item on UaExpert, many errors appears on Log window. The monitored value is refreshed correctly. I tried with UaExpert 1.5.1 and 1.6.3 : same result. LibUA is the latest version, downloaded today, NET trunk ( .NET Framework 4.8 ). The OS is Windows 10 22H2 
[image]
.

stefanofraccaro, 2023-03-02:
[LibUA-Log.log](https://github.com/nauful/LibUA/files/10870682/LibUA-Log.log)

stefanofraccaro, 2023-03-02:
Maybe I have found the problem, but I don't remember to have never activated these settings. To reproduce the problem, go to Settings -> Configure UaExpert... and set Trace.StackTraceEnabled=true and Trace.SDKTraceEnabled=true
If you revert these settings to its default value (=false) the communication seems fine.

nauful, 2023-03-03:
Hello,

This is a known warning which I don't plan to fix because it doesn't impact functionality.

More details: acknowledgements and publish notifications are calculated asynchronously so their counts don't line up immediately. I could make this synchronous, which would fix that warning but performance would decrease significantly when publishing many monitored items to many different clients.

closed 2023-03-03 by nauful