nauful/LibUA #119: Error with ExpandedNodeId

mosseedwin, 2023-04-28:
Generated exception when ExpandedNoId has server index.

nauful, 2023-04-30:
Can you share a Wireshark capture of this case?

mosseedwin, 2023-05-01:
> ###
> [Apr 30, 2023](#issuecomment-1529011691)

LibUA used as OPC client.

[opcua.zip](https://github.com/nauful/LibUA/files/11368214/opcua.zip)

nauful, 2023-05-02:
Sorry, I haven't implemented support for `ExpandedNodeId`. To add that, you might implement `VariantCodingSize`, `VariantEncode` and `VariantDecode` here and the lines under that:

https://github.com/nauful/LibUA/blob/8462ab98c820c87a82e8b0a1e418eb6365c7fb01/NET%20Core/LibUA/Encoding.cs#L967

Otherwise I will add this to my research list and implement this type in the future.

closed 2023-05-02 by nauful