nauful/LibUA #120: client.Write method is not working for me

Sanjay-pujari, 2023-06-01:
I have created a console application where I'm using client.Write method to write new value to required node but no change in value and getting status code "2155085824" in response.

[image]

nauful, 2023-06-01:
This error corresponds to the enum for BadTypeMismatch. Might need unsigned int, long, a float or a double depending on the type of the node. Check the data type for the node on the server.

Closing, not a library issue.

closed 2023-06-01 by nauful

Sanjay-pujari, 2023-06-06:
I have created another node of type int and just to assign 6, but still not writing value to node, where as when tried with OPC UA client then it is writing the value

uint[] respStatuses;
                //client.SetPublishingMode(true);
                client.Write(new WriteValue[]
                    {
                        new WriteValue(
                            new NodeId(2, "SiemensPLCs.PLC1500.HMI.Station_Status[2]"), NodeAttribute.Value,
                            null, new DataValue(6, StatusCode.GoodClamped, DateTime.Now))
                    }, out respStatuses);

This is the code snippet

Sanjay-pujari, 2023-06-06:
> This error corresponds to the enum for BadTypeMismatch. Might need unsigned int, long, a float or a double depending on the type of the node. Check the data type for the node on the server.
> 
> Closing, not a library issue.

Hello Nauful, thank you for the response.
I have double verified the data type and value assigned but still facing issue. please check my comment with more details

nauful, 2023-06-06:
The UA data type is `Word`, which in C# is `ushort`. Sending 6 as a constant is an `int` (`Int32` for UA) Try declaring a variable with that type specified:

```
ushort value = 6u;
... new DataValue(value, ...)
```

nauful, 2023-06-06:
Try `StatusCode.Good` instead of `StatusCode.GoodClamped`. If that doesn't work, I need more info:

With an unencrypted connection (security `None` or `Sign` but not `SignAndEncrypt`), can you take a Wireshark capture of writing the value from your OPC client, then of writing your value from this test app?

Sanjay-pujari, 2023-06-07:
I have tried with security none, but in this case I'm event not able to read the value as well. But when I'm trying with SignAndEncrypt, I'm able to read the value but can't write. When I checked the status code returned by "write" method then it says

BadWriteNotSupported = 0x80730000, // The server not does support writing the combination of value

But I'm passing the correct

nauful, 2023-06-07:
Sounds good. Your server may have been rejecting the additional parameters.