nauful/LibUA #121: Issue

AnyelaMolina, 2023-06-01:
On the server when Basic256 security types are configured with Sign and with SignAndEncrypt, the client cannot connect.
Beforehand thank you very much.

nauful, 2023-06-02:
Can you take a Wireshark capture?

I would guess the server is rejecting the client's certificate unless you've already added the client certificate to the server's trusted certificate store.

AnyelaMolina, 2023-06-02:
Excellent day, here I leave the capture
[Basic 256 SignAndEncrypt.pdf](https://github.com/nauful/LibUA/files/11636161/Basic.256.SignAndEncrypt.pdf)
[Basic 256 Sing.pdf](https://github.com/nauful/LibUA/files/11636163/Basic.256.Sing.pdf)

AnyelaMolina, 2023-06-02:
thank, I saw my mistake

nauful, 2023-06-05:
Sounds good!

closed 2023-06-05 by nauful

thousand3, 2023-06-07:
Hi AnyelaMolina, is it possible to share your solution/mistake? Because i'm facing the same issue here. 

Btw, When use LibUA as server, where is the server's trusted certificate store folder?

nauful, 2023-06-07:
If you're using LibUA as a server,
1. Your client needs to trust the server certificate.
2. The server doesn't use a certificate store by default. You can implement that here to accept/reject client certificates (the sample server accepts all by default): https://github.com/nauful/LibUA/blob/8462ab98c820c87a82e8b0a1e418eb6365c7fb01/NET%20Core/TestServer/Program.cs#L113

You can check that `byte[] clientCertificate` is a trusted certificate (content exists in some trusted folder) if you want to.

thousand3, 2023-06-08:
Got it. Let me test it out. Thanks for your quick response. Appreciate it.