nauful/LibUA #124: No event notifications

brotleuchte, 2023-07-01:
Hi,
I'm using the nuget package and trying to get event notifications to work by using the client sample code. The UAExpert client shows events, e.g. of type ProgramTransition or AuditEvent. I also tried the Server Object Node.

```
statusCode = m_client.CreateMonitoredItems(m_subscriptionId, TimestampsToReturn.Both,
    new MonitoredItemCreateRequest[]
    {
         new MonitoredItemCreateRequest(
             new ReadValueId(new NodeId(5, 31), NodeAttribute.EventNotifier, null, new QualifiedName()),
             MonitoringMode.Reporting,
             new MonitoringParameters(clientHandleEventMonitor, 0, new EventFilter(eventFilterOperands, null), 1, true)),
    }, out monitorCreateResults);
```

Am I missing something such as no AuditEvent notifications by default or so?

nauful, 2023-07-03:
I would check:
1. The target node is correct and publishes.
2. Monitoring mode is correct in the server.
3. `eventFilterOperands` are correct.

This should trigger your application's `NotifyEventNotifications` method.

Could you please share a Wireshark traffic capture with no security mode? This will help me troubleshoot further.

brotleuchte, 2023-07-05:
Hey nauful, 
I think my eventFilterOperands are not correct. Unfortunately other things got prioritized and everything else is working so far so I can't spend much time to this event topic the next couple of days. But it is still on my list. I appreciate your work so =)